realsenseai/librealsense #10506: D455 USB permissions issue on WSL2

tomstix, 2022-05-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu 20.04 on WSL2  | 
| Kernel Version (Linux Only)    |  5.10.102.1-microsoft-standard-WSL2 | 
| Platform                                 | PC |
| SDK Version                            |libuvc installation | 
| Language                            |  C++                         | 
| Segment			|  Robot | 

### Issue Description
Hello,

I tried installing the SDK using the libuvc installation script on Windows Subsystem for Linux. The camera is connected via usbip. This works but I have to use sudo for realsense-viewer or other tools to be able to connect to the camera. When I run realsense-viewer without sudo, it doesn't connect to the camera and the console output is:
`13/05 10:33:07,336 ERROR [140169059489536] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 13/05 10:33:07,337 ERROR [140169152157440] (sensor.cpp:572) acquire_power failed: failed to set power state
 13/05 10:33:07,339 WARNING [140169152157440] (rs.cpp:310) null pointer passed for argument "device"
 13/05 10:33:07,339 WARNING [140169152157440] (rs.cpp:2691) Couldn't refresh devices - failed to set power state`
 
I added 99-realsense-libusb.rules to /etc/udev/rules.d/ and restarted the udev service but I still can't connect without sudo. My user is in the plugdev group.
Any suggestions on how to fix this?

MartyG-RealSense, 2022-05-13:
Hi @tomstix  We do not receive many WSL2 questions as it is my understanding that it does not have support for USB devices by default.  It sounds as though you are performing a workaround for that with usbip.  Is that correct, please?

Assuming that the problem is related to not having permissions without using sudo. A RealSense user at the link below offers advice about reloading udev after the /etc rule is added.

```
sudo udevadm control --reload-rules
sudo udevadm trigger
```

https://stackoverflow.com/questions/62134563/how-to-give-permission-to-intel-realsense-camera-on-ubuntu/62369143#62369143

tomstix, 2022-05-13:
Hello @MartyG-RealSense, thank you for your quick response.
Microsoft recently added support for USB/IP to their WSL2-Kernel. See https://docs.microsoft.com/en-gb/windows/wsl/connect-usb.

I tried to connect the camera to a laptop running Windows 11 and one on Windows 10.
On the Windows 11 one I actually just got it working after fiddling around with the udev rules. I actually didn't change anything in the end but I think completely unplugging and replugging the camera and also restarting WSL finally did the trick. However, on the Windows 10 machine I still have the permissions issue.
But as I realized that this is rather a WSL-issue I think we can close it here.

closed 2022-05-13 by tomstix

MartyG-RealSense, 2022-05-14:
Thanks very much @tomstix for the update and for sharing the details of your solution with the RealSense community!

gonultasbu, 2022-06-17:
Hello @tomstix, I have tried setting up the realsense stack on WSL2, however the biggest problem for me has been the lack of extra kernel modules `videodev, uvcvideo, hid-sensor-accel-3d, hid-sensor-gyro-3d` for the WSL2 linux kernel. Did you manage to solve this problem? I believe without those, realsense-viewer is unable to detect the rs cameras.

Sri1396, 2022-08-25:
> Hello @MartyG-RealSense, thank you for your quick response. Microsoft recently added support for USB/IP to their WSL2-Kernel. See https://docs.microsoft.com/en-gb/windows/wsl/connect-usb.
>
> I tried [...]

Hey @tomstix, I run into the same issue as well, when I use sudo for realsense-viewer it works but does not work otherwise. and I have WSL2 setup too. What did you actually mean by fiddling around with the udev rules? Could you clarify please?

may-iCOMgiken, 2024-08-01:
> Hello @tomstix, I have tried setting up the realsense stack on WSL2, however the biggest problem for me has been the lack of extra kernel modules `videodev, uvcvideo, hid-sensor-accel-3d, [...]
hello @gonultasbu I also got the same issue. the lack of the videodev, ucvideo. Do you already found the solution. If so, may you share the solution?

gonultasbu, 2024-08-01:
@may-iCOMgiken I was not able to find a solution.