realsenseai/librealsense #10507: self-calibrate

L-xn, 2022-05-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Intel RealSense Viewer | 
| Operating System & Version |   {Win (10)  | 

I am using the Intel RealSense Viewer to self-calibrate.But an error message appears：invalid value in rs2_run_on_chip_calibration_cpp hwmon command 0x80(8108) failed.I want to ask for help.
[image]
[image]

MartyG-RealSense, 2022-05-13:
Hi @L-xn  When you expand open the **Stereo Module** options, does it only have the option to enable Depth?  There should also be 'Infrared' and 'Infrared 2' options in that section.  If one or both of the Infrared options are missing then you may need to completely uninstall the two 'Depth' and 'RGB' RealSense camera drivers in the Windows **Device Manager** interface in order for on-chip calibration in the Viewer to work correctly.

This problem can occur if the RealSense **Universal Windows Platform (UWP)** driver has been installed on a computer, as described at the discussion at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/4419989666323-Installing-D400-UWP-driver-disables-Infrared-streams-on-D435

Within that discussion, instructions for the Device Manager driver uninstall are here:

https://support.intelrealsense.com/hc/en-us/community/posts/4419989666323/comments/4431239847443

L-xn, 2022-05-13:
> Hi @L-xn When you expand open the **Stereo Module** options, does it only have the option to enable Depth? There should also be 'Infrared' and 'Infrared 2' options in that section. If one or both of [...]

I turned on the infrared switch as you said, but the result was still the same as the above, there was an error.  
[image]

MartyG-RealSense, 2022-05-14:
Thank you.  Next, please try resetting your camera's calibration values to their factory-new defaults in the Viewer using the procedure at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

closed 2022-05-26 by MartyG-RealSense

MartyG-RealSense, 2025-08-07:
Thanks so much @jstmn for sharing what worked for you in your own situation!