realsenseai/librealsense #10508: types.h:898 - Depth stream start failure

deepak-asink, 2022-05-13:
Getting depth stream failure frequently on realsense viewer with D435i camera.

MartyG-RealSense, 2022-05-14:
Hi @deepak-asink  Is the camera ever able to stream depth or does the error only happen sometimes?

When it does happen, does unplugging the camera from the USB port and plugging it back in correct the problem?

deepak-asink, 2022-05-16:
Yes, the error is happening more frequently. After plugging it back it works only sometimes.

MartyG-RealSense, 2022-05-16:
Please provide information about the questions below.

Which version of the RealSense Viewer are you using (for example, 2.50.0). 

Are you using the Windows or Linux operating system on your computer?

If you are using Linux, what is your kernel version?

deepak-asink, 2022-05-16:
version 2.50.0
Ubuntu 20.04
kernel 5.13.0-41-generic

MartyG-RealSense, 2022-05-16:
The RealSense SDK does not currently officially support kernel 5.13.  It supports 5.4 if building from Debian packages or 5.8 and 5.11 if building from source code.  Whilst unsupported kernels can work with the SDK, there may be unpredictable consequences in regard to stability.

Alternatively, you can bypass the kernel by building the SDK from source code with the **-DFORCE_RSUSB_BACKEND=true** flag included in the CMake build instruction.  With an RSUSB build, the SDK is not dependent on Linux versions or kernel versions and does not require patching.  

This is a useful mode if you have kernel related conflicts but your project cannot change the kernel version or you are using an exotic hardware / software configuration such as an industrial computer that may have problems with a conventional SDK build.

deepak-asink, 2022-05-17:
I have another D435i camera which is working completely fine. Also I tried to use the same camera on raspberry pi and windows, but getting the same problem there also.
I am already using  -DFORCE_RSUSB_BACKEND=true flag while building.

MartyG-RealSense, 2022-05-17:
Thank you for the information.  Could you check whether resetting the affected D435i's calibration to its factory-new default calibration settings in the RealSense Viewer using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 resolves the **Depth stream start failure** error, please?

deepak-asink, 2022-05-21:
I tried the solution suggested by you, but it didn't fixed the issue.
Can you provide details of how to claim warranty ?

MartyG-RealSense, 2022-05-21:
If you purchased the camera from the official RealSense Store then you can contact the Store to discuss return using the contact form at the link below, selecting **Return / Warranty** as the topic.

https://www.intel.com/content/www/us/en/forms/support/intel-realsense-store-inquiry.html

If you purchased the camera from a retailer other than the RealSense Store then please contact that retailer to find out their **Return Material Authorization (RMA)** returns procedure.

MartyG-RealSense, 2022-06-02:
Hi @deepak-asink  Do you have an update that you can provide about your wish to carry out a camera return, please?  Thanks!

deepak-asink, 2022-06-02:
I couldn't find the stocking ID of the device, can you help me with that.

MartyG-RealSense, 2022-06-02:
The D435i has a manufacturer product ID of 82635D435IDK5P

RealSense cameras have a return period of 30 days after delivery and a warranty period of 1 year.

[image]

closed 2022-06-13 by MartyG-RealSense