realsenseai/librealsense #10511: D435 camera repeatedly crashes

r91andersson, 2022-05-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.12.07.150 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.253-tegra                                         | 
| Platform                                 | NVIDIA Jetson NX |
| SDK Version                            |  v2.3.1                           | 
| Language                            |  C                           | 
| Segment			|  Robot                          | 

### Issue Description
After launching the camera, it takes approx. 1 min before the camera crashes. The first error I can see is this (but there are several of them)
`(librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device`

After the device is recognized as disconnected, it finds the camera once again and then looses the connection all over again.
I actually have two cameras attached to the NX card. But only one of them fails. The otherD435 camera that works has the exactly same firmware and config.

Here is the full log:
```
process[camera5/realsense2_camera-2]: started with pid [30092]
[ INFO] [1652515685.845323547]: Loading nodelet /camera5/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:
[ INFO] [1652515685.854158001]: Initializing nodelet with 4 worker threads.
[ INFO] [1652515686.463497743]: RealSense ROS v2.3.1
[ INFO] [1652515686.463636624]: Built with LibRealSense v2.50.0
[ INFO] [1652515686.463749744]: Running with LibRealSense v2.50.0
[ INFO] [1652515686.526910319]:  
[ INFO] [1652515686.543148344]: Device with serial number 138422076624 was found.

[ INFO] [1652515686.543326137]: Device with physical ID /sys/devices/3610000.xhci/usb2/2-3/2-3.4/2-3.4.1/2-3.4.1:1.0/video4linux/video0 was found.
[ INFO] [1652515686.543516761]: Device with name Intel RealSense D435 was found.
[ INFO] [1652515686.545360190]: Device with port number 2-3.4.1 was found.
[ INFO] [1652515686.661149281]: Device with serial number 135122072080 was found.

[ INFO] [1652515686.661269698]: Device with physical ID /sys/devices/3610000.xhci/usb2/2-3/2-3.4/2-3.4.2/2-3.4.2:1.0/video4linux/video3 was found.
[ INFO] [1652515686.661347778]: Device with name Intel RealSense D435 was found.
[ INFO] [1652515686.663124934]: Device with port number 2-3.4.2 was found.
[ INFO] [1652515686.663247623]: Device USB type: 3.2
[ INFO] [1652515686.671044250]: getParameters...
[ INFO] [1652515687.071148362]: setupDevice...
[ INFO] [1652515687.071624427]: JSON file is not provided
[ INFO] [1652515687.071873292]: ROS Node Namespace: camera5
[ INFO] [1652515687.072180492]: Device Name: Intel RealSense D435
[ INFO] [1652515687.073149423]: Device Serial No: 135122072080
[ INFO] [1652515687.073675056]: Device physical port: /sys/devices/3610000.xhci/usb2/2-3/2-3.4/2-3.4.2/2-3.4.2:1.0/video4linux/video3
[ INFO] [1652515687.073820880]: Device FW version: 05.12.07.150
[ INFO] [1652515687.073919217]: Device Product ID: 0x0B07
[ INFO] [1652515687.074566962]: Enable PointCloud: Off
[ INFO] [1652515687.075150324]: Align Depth: On
[ INFO] [1652515687.076245847]: Sync Mode: On
[ INFO] [1652515687.077150073]: Device Sensors: 
[ INFO] [1652515687.087769972]: Stereo Module was found.
[ INFO] [1652515687.101116661]: RGB Camera was found.
[ INFO] [1652515687.101275766]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1652515687.103657788]: num_filters: 1
[ INFO] [1652515687.104038557]: Setting Dynamic reconfig parameters.
[ WARN] [1652515687.268213818]: Param '/camera5/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1652515687.298262758]: Done Setting Dynamic reconfig parameters.
[ INFO] [1652515687.302486864]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1652515687.306276538]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1652515687.311120774]: setupPublishers...
[ INFO] [1652515687.320789886]: Expected frequency for depth = 30.00000
[ INFO] [1652515687.424146755]: Expected frequency for color = 30.00000
[ INFO] [1652515687.501777414]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1652515687.581456143]: setupStreams...
[ INFO] [1652515687.587005181]: insert Depth to Stereo Module
[ INFO] [1652515687.587359742]: insert Color to RGB Camera
[ INFO] [1652515687.601952386]: SELECTED BASE:Depth, 0
[ INFO] [1652515687.655041800]: RealSense Node Is Up!
[ WARN] [1652515687.727135262]: 
[... 53 lines cut ...]
[ERROR] [1652515863.136561881]: The device has been disconnected!
 14/05 10:11:03,140 ERROR [546972762496] (librealsense-exception.h:52) xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
 14/05 10:11:03,140 ERROR [546972762496] (sensor.cpp:596) release_power failed: xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
 14/05 10:11:03,140 ERROR [546972762496] (librealsense-exception.h:52) xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
 14/05 10:11:03,140 ERROR [546972762496] (librealsense-exception.h:52) xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
 14/05 10:11:03,141 ERROR [546972762496] (sensor.cpp:596) release_power failed: xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
 14/05 10:11:03,257 ERROR [546972762496] (librealsense-exception.h:52) acquire: Cannot open '/dev/video3 Last Error: No such file or directory
[ERROR] [1652515863.260743694]: Exception: acquire: Cannot open '/dev/video3 Last Error: No such file or directory
 14/05 10:11:03,259 ERROR [546972762496] (librealsense-exception.h:52) xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
 14/05 10:11:03,259 ERROR [546972762496] (sensor.cpp:596) release_power failed: xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
 14/05 10:11:03,418 ERROR [546727518592] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 14/05 10:11:03,419 ERROR [546727518592] (librealsense-exception.h:52) xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
 14/05 10:11:03,419 ERROR [546727518592] (sensor.cpp:596) release_power failed: xioctl(VIDIOC_REQBUFS) failed Last Error: No such device
 14/05 10:11:03,419 ERROR [546727518592] (error-handling.cpp:78) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
[ INFO] [1652515864.393973678]:  
[ INFO] [1652515864.403868198]: Device with serial number 138422076624 was found.

[ INFO] [1652515864.404508584]: Device with physical ID /sys/devices/3610000.xhci/usb2/2-3/2-3.4/2-3.4.1/2-3.4.1:1.0/video4linux/video0 was found.
[ INFO] [1652515864.404795432]: Device with name Intel RealSense D435 was found.
[ INFO] [1652515864.406917774]: Device with port number 2-3.4.1 was found.
[ERROR] [1652515864.411232696]: The requested device with serial number 135122072080 is NOT found. Will Try again.

```

I've tried to reboot the camera and also set the intial_reset to true, but that results in once the camera restarts, the driver cant find the camera.

MartyG-RealSense, 2022-05-14:
Hi @r91andersson  the first point that I notice in your log is that you have a mismatch between your RealSense ROS wrapper version and your librealsense SDK version.  It is important to match the wrapper version as closely as possible with the librealsense version recommended for a particular wrapper version in its release notes.

If you are using ROS1 (Kinetic, Melodic, Noetic) then the correct match for librealsense SDK 2.50.0 is **2.3.2**.  You have wrapper 2.3.1 installed, which is designed to be matched with librealsense 2.48.0.

MartyG-RealSense, 2022-05-16:
If you previously installed librealsense from packages and want to uninstall it, information for doing so is at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

At https://github.com/IntelRealSense/librealsense/issues/5773#issuecomment-581477785 a RealSense team member advises how to install a specific librealsense version from packages.

MartyG-RealSense, 2022-05-16:
There is a known issue with severe lag when using align_depth = true on Nvidia Jetson boards specifically.  The issue does not occur on non-Jetson computers and computing devices.  Pointcloud generation has similar problems.  The problems are discussed at https://github.com/IntelRealSense/realsense-ros/issues/2334 and https://github.com/IntelRealSense/realsense-ros/issues/1967

Using librealsense 2.43.0 and ROS1 wrapper 2.2.23 is typically a successful solution.

MartyG-RealSense, 2022-05-17:
As your firmware is newer than 5.12.8.200 (which has a minimum SDK version of 2.39.0), your current firmware should be back-compatible with SDK 2.43.0.

r91andersson, 2022-05-18:
@MartyG-RealSense Thanks

closed 2022-05-18 by r91andersson