realsenseai/librealsense #10512: RealSense Viewer stops working wit D405 after 10 seconds

ashenkin, 2022-05-14:
Hi all,

The RealSense Viewer (v 2.50.0) displays full color 3D imagery from the D405 for about 10 seconds, before flickering out.  If I turn off post-processing, I can see the non-color depth map in 3D.  If I close and restart the viewer, all is well again for 10 seconds.  Any thoughts?

Thanks.

Camera: D405
Firmware: 05.13.00.50
OS: Win 10 Pro Build 19044

MartyG-RealSense, 2022-05-14:
Hi @ashenkin  If you are using the 3D point cloud mode of the RealSense Viewer and the point cloud flickers and disappears after about 10 seconds, it is typically caused by the Viewer's two **GLSL** options ('Rendering' and 'Processing') being enabled in its settings interface.  If they are disabled in the settings then the problem usually disappears immediately.  Instructions for disabling GLSL can be found at https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

MartyG-RealSense, 2022-05-21:
Hi @ashenkin  Do you require further assistance with this case, please? Thanks!

ashenkin, 2022-05-22:
All set - thank you!

MartyG-RealSense, 2022-05-22:
Thanks very much @ashenkin for the update!   As you do not require further assistance, I will close this case.  Thanks again!

closed 2022-05-22 by MartyG-RealSense