realsenseai/librealsense #10513: The tool ( Intel RealSense Viewer v2.50.0 ) can't read the device data stream

timeparticle, 2022-05-15:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|
The camera have connected to my PC , but the tool ( Intel RealSense Viewer v2.50.0 ) can't find  the device as follow picture shows
[image]

MartyG-RealSense, 2022-05-15:
Hi @timeparticle  Could you first try completely uninstalling and re-installing the RGB and Depth drivers from the Device Manager using the instructions in the link below to see whether it corrects your camera detection problem, please?  Thanks!

https://support.intelrealsense.com/hc/en-us/community/posts/4419989666323/comments/4431239847443

timeparticle, 2022-05-15:
Hi Marty G ,Thank you for your prompt reply .
I have tried the above process you recommended，but it didn't work.

MartyG-RealSense, 2022-05-15:
Thank you for trying the uninstall-reinstall procedure.

Next, please check in the Windows privacy settings whether permission to access the camera hardwae and permission for applications to use the camera are oth set to On using the instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

timeparticle, 2022-05-16:
Thanks a lot,I have found the problem.，my Windows OS in a special version  does not have permission to open the camera.

timeparticle, 2022-05-16:
> Thanks a lot，My Windows OS in a special version does not have permission to open the camera, then I use the Group Strategy open the camera permission and restart my computer ,but it didn't work.

I use the system Group Strategy ,open all camera-related permissions ,and then the problem was inexplicably solved.

MartyG-RealSense, 2022-05-16:
It's excellent to hear that you achieved a solution.  I will therefore close this case.  Thanks again!

closed 2022-05-16 by MartyG-RealSense