realsenseai/librealsense #10518: Frame set not supported - Qt / QML

Amayess, 2022-05-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D455} | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   {Win (10) | 
| Kernel Version (Linux Only)    |  ()                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { v2.50.0 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot}                          | 

### Issue Description

Hi,

I am trying to reproduce this project **https://github.com/Jiloc/tighty** with following all the steps. I was able to get the application work, i.e. the interface is displayed with all the Qt components without any problems, but when I click on the "start button" to start streaming the intel scanner, I get an error saying "**Frame set not supported yet**"...

Can you help me to solve it please ? 

Ps : I use Visual studio 2019 ,Qt 5.15 ( Qml not Qtwidget) and intel RealSense D455

MartyG-RealSense, 2022-05-16:
Hi @Amayess  It looks as though the **jiloc/tighty** project was designed with installation on MacOS in mind, though the information at the top of this case states that you are using a PC and Windows.

Assuming that the project is installable on Windows, its **3rdParty** folder contains an extremely old version of the librealsense SDK (version 2.9.1 dating to February 2018).  The D455 camera model was not supported by librealsense until two years later in 2020's version 2.38.1.  So your D455 could not function with this project in its default state and you may need to replace the bundled 2.9.1 version in the 3rdParty folder with at least librealsense 2.38.1 (the current librealsense version is 2.50.0).

**2.38.1**
https://github.com/IntelRealSense/librealsense/releases/tag/v2.38.1

**2.50.0**
https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

MartyG-RealSense, 2022-05-16:
I suspect that it might be looking for a **built** version of librealsense rather than the raw unbuilt source code folder.  Therefore, I would speculate that the approach below may be worth trying.

1.  Download the  **Intel.RealSense.SDK-WIN10** installer file for the binary version of the 2.50.0 SDK from the 'Assets' file list of the SDK releases page and run the installer file to install the full SDK for Windows on your computer.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

[image]

2.  Navigate to the SDK folder at the Windows location **C: > Program Files (x86)** and find the folder called **Intel RealSense SDK 2.0**

[image]

3.  Rename the Intel RealSense SDK 2.0 folder to **librealsense** and move it to the Jiloc example's **3rdParty** folder in place of the original librealsense folder for 2.9.1.

[image]

If you open the newly renamed built 'librealsense' folder for 2.50.0 then you can see that it has similar elements to the original 2.9.1 librealsense folder such as bin, include and lib.

[image]

Amayess, 2022-05-17:
Hi @MartyG-RealSense 

I was able to solve the display problem but now I have a problem with the image format, when I select the RBG_888 format I have an absolutely unclear image (see image below)

What is the best image format to view my stream? 

[image]

MartyG-RealSense, 2022-05-17:
In the RealSense SDK, depth is Z16 and color is RGB8.  What choices does Qt offer you for the format, please?

Amayess, 2022-05-17:
All the choices are on this link : https://doc.qt.io/qt-5/qimage.html#Format-enum 

What is the best choice, please ?

MartyG-RealSense, 2022-05-17:
How does the image look if set to **RGB16**

Amayess, 2022-05-17:
When i set RBG16, i get something like this ! not that bad but actually I want a similar result to the intel realsense viewer
[image]

MartyG-RealSense, 2022-05-17:
RealSense RGB color uses RGB8 by default but there does not seem to be an 8-bit RGB color option in Qt, with the closest to it probably being a monochrome **Grayscale8**

Amayess, 2022-05-17:
With the Grayscale8 it's even worst :/ Nothing that can be understood 
[image]

MartyG-RealSense, 2022-05-17:
Ok, let's assume for the moment that RGB16 is the best RGB format that you can get for Qt.

Are you using a depth image too?

Amayess, 2022-05-17:
Yes am using it ! Acutally this project https://github.com/Jiloc/tighty is only made for the depth camera so yeah am using the depth camera

MartyG-RealSense, 2022-05-17:
Ah I see, I thought you were using the camera's 8-bit RGB image.  The depth image of the camera is 16 bit, so it makes sense that the RGB16 image would resemble the RealSense depth image a little.

You are unlikely to achieve a depth image in Qt that resembles the Viewer, as the Viewer applies a range of post-processing and colorization filters to the image by default.  If the image was being generated in a script that you had created yourself then it would look more like the Qt one above.

Amayess, 2022-05-17:
Even if i remplace the main program with the "Capture-example" ? i still can't get the same colors ?

MartyG-RealSense, 2022-05-17:
I honestly do not know enough about Qt to advise on the possibilities of what could be achieved, unfortunately.

Amayess, 2022-05-18:
Hi @MartyG-RealSense 

I made it ! thank you for your help 👍

Amayess, 2022-05-18:
HI @MartyG-RealSense 

I have a question, is it possible to "export to ply file" a 2D depth camera stream ?

MartyG-RealSense, 2022-05-18:
.ply is a point cloud data format so a 2D depth image cannot be exported to it.  If you need to preserve depth information in the file, a **.raw** format image would be a good choice.

At https://github.com/IntelRealSense/librealsense/issues/1485#issuecomment-379703868 a RealSense team member provides a C++ script for doing so.

MartyG-RealSense, 2022-05-18:
Yes, create a C++ script to generate a point cloud and export to ply.  https://github.com/IntelRealSense/librealsense/issues/6411 has a simple script for doing so.

Amayess, 2022-05-19:
Hi @MartyG-RealSense 

I am able to generate the point cloud but i have a little question when the function **"export_to_ply"** is called where can i find the exported file ? 

thank you

MartyG-RealSense, 2022-05-19:
I would expect the file to be generated in the same folder location that the script is launched in.  A Windows user of export_to_ply who also had a missing ply seems to suggest in https://github.com/IntelRealSense/librealsense/issues/6979#issuecomment-696125446 that in their case it was running the built .exe instead of using Visual Studio debug mode that got it working.

MartyG-RealSense, 2022-05-19:
Would it be possible to create a **QML button** and trigger the export_to_ply instruction when the button is clicked?

https://doc.qt.io/qt-5/qml-qtquick-controls2-button.html

Amayess, 2022-05-19:
I already found a way, thank you ! have a good day :)

closed 2022-05-25 by MartyG-RealSense