realsenseai/librealsense #10520: How can I know the absolute error over the different distance about D455 in ROS.

jiejie567, 2022-05-17:
At first, thank you for the great cameras  making my work better.
I really need the information how does the error change while the depth value changes. Does it conform to the linear model or the quadratic function model if only considering the depth.

MartyG-RealSense, 2022-05-17:
Hi @jiejie567  With RealSense 400 Series cameras, error starts at around zero at the camera position and increases linearly over distance.  This is known as **RMS Error**.  Point 5 of the section of Intel's camera tuning guide linked to below provides further information about RMS error.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-verify-performance-regularly-on-a-flat-wall-or-target

jiejie567, 2022-05-21:
@MartyG-RealSense Thank you for your detailed reply. I wonder  is there a bias error except RMS error? Does the RMS error includes the bias?

MartyG-RealSense, 2022-05-21:
Bias is referenced in regard to depth accuracy in Intel's on-chip calibration guide, which states that depth bias can be removed with a **Tare** calibration (which improves ground-truth measurement accuracy).

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#section-3-depth-accuracy

Section D of the section of the same guide linked to below also discusses potential causes of bias and noise in the ground truth value.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#section-usage-guidelines-and-considerations

My understanding is that bias is not directly included in the RMS value, but there can be bias in the ground truth that can be corrected with a Tare calibration.

jiejie567, 2022-05-24:
Thank you, MartyG. I can learn a lot from your reply. But I still have a question. Take the D455 as an example, if  the error starts at around zero at the camera position and increases linearly over distance, why the document only mentions the error less than 2% at 4 meters rather than less than 2% at different distance?

MartyG-RealSense, 2022-05-24:
The error percentage values provide a theoretical estimate of what can be expected to be achieved at a certain distance with a particular camera model.  The differences between models are compared in a chart at the link below.

https://www.intelrealsense.com/compare-depth-cameras/

A difference between the D455 and D435 / D435i is that the D455 has 2x the depth measurement accuracy over distance.  Typically on 400 Series camera models, RMS depth error starts becoming noticeable at the 3 meter distance and beyond.  With the D455 though, because of the 2x improvement in RMS error it has the same amount of error at 6 meters distance that the other 400 Series models have at only 3 meters, meaning that the D455 can depth sense more accurately as distance increases compared to D435, D435i, etc.

MartyG-RealSense, 2022-05-30:
Hi @jiejie567  Do you require further assistance with this case, please?  Thanks!

jiejie567, 2022-05-30:
Thank you for your help,MartyG @MartyG-RealSense ,your information is important to me. I will use L515 temporarily, whose accuracy doesn't differ much by the distance.

MartyG-RealSense, 2022-05-30:
You are very welcome, @jiejie567 

When you are satisifed with the outcome of this case, please feel free to close it using the 'Close Issue' button under the comment writing box.  Good luck!

closed 2022-05-30 by jiejie567