realsenseai/librealsense #10525: how to tune the IR emitter power? dynamically in ROS2-foxy

cheukwaylee, 2022-05-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2022-05-18:
Hi @cheukwaylee  May I first ask if you are using the RealSense **ros2** wrapper (version numbers up to 3.2.2) or the **ros2_beta** wrapper (such as version 4.0.4).  Thanks!

cheukwaylee, 2022-05-18:
yes, i am using this wrapper "https://github.com/IntelRealSense/realsense-ros/tree/ros2" which can set the d435i parameter in "rs_launch.py", but idk how i can tune the laser power ( my project is about SLAM by using d435i in low-light environment, but the IR pattern makes the ORB feature extractor confusing). THANKS!

cheukwaylee, 2022-05-18:
in ROS1-noetic, rqt_reconfigure can be used to tune the parameters including laser power, but i cannot find similar function in ROS2

MartyG-RealSense, 2022-05-18:
ROS2 uses a different parameter system to ROS1's rqt_reconfigure / dynamic_reconfigure.

In the **ros2** wrapper you can still access the graphical user interface of rqt_reconfigure to make real-time canges to settings with the command **ros2 run rqt_reconfigure rqt_reconfigure** after launch has been completed.

https://github.com/IntelRealSense/realsense-ros/tree/ros2#view-and-modify-camera-controls-params-in-runtime

Alternatively, https://github.com/IntelRealSense/realsense-ros/issues/1379 provides information about setting the value of the laser_power parameter in the **ros2** wrapper using a **.yaml** file.

cheukwaylee, 2022-05-18:
i tried "ros2 run rqt_reconfigure rqt_reconfigure" in ROS2, but i can NOT see a tune-able parameter about laser power. (i also tried in ROS1-noetic where i can see the laser power tunner)

MartyG-RealSense, 2022-05-18:
Is there an **emitter_enabled** option in ROS2 rqt_reconfigure that you could change?  Disabling the emitter would have the same effect of removing the dot pattern as minimizing the value of laser_power does.  I believe that emitter_enabled should be in the **stereo_module** section of rqt_reconfigure.

cheukwaylee, 2022-05-18:
ok, i can find ".emitter_enabled" and ".laser_power" both in "depth_module", so now they both can be tunned in ROS2-foxy rqt_reconfigure. THANKS :)

as for my real application, i dont wanna completely turn off the IR laser, cuz in the low-light environment, i guess some IR can help with visibility, but i dont wanna the dot pattern confuses the ORB feature point extractor, it is a conflict i need to solve.

MartyG-RealSense, 2022-05-18:
400 Series cameras can make use of any visible light in a scene to assist with depth analysis, so it is possible to use an external patternless **IR illuminator** lamp in close proximity to the camera in low-light / dark scenes, as described at https://github.com/IntelRealSense/librealsense/issues/2000

When adjusting Laser Power, the projector should stay on if the Laser Power is '1' but the dot pattern will be invisible to the camera.

cheukwaylee, 2022-05-19:
instead of using an external patternless IR illuminator lamp, I would like to see if it is possible to remove the IR white dot pattern if the pattern is provided by manufacturer as a part of the datasheet. because i suppose the pattern can be considered as the upper layer adding directly to the real scene.

MartyG-RealSense, 2022-05-19:
The D415 camera model supports using a camera configuration preset called **Left Imager Color w/o IR Pattern** that can remove the dot pattern by adjusting the camera's **color correction matrix** settings.

https://dev.intelrealsense.com/docs/d400-series-visual-presets#section-preset-table

Further information about this can be found in Point 3 of the section of Intel's camera tuning guide linked to below.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-use-the-left-color-camera

MartyG-RealSense, 2022-05-19:
I just wanted to highlight that the RealSense color correction matrix technique for dot pattern removal exists even if it is not suited to the D435i model, in case that changing camera model was a possibility for your project.

I would reiterate my advice above to try setting Laser Power to '1' so that the pattern disappears but the projector remains on.

Alternatively, try enabling **emitter_on_off** in ROS so that the pattern alternates on and off each frame.

cheukwaylee, 2022-05-20:
maybe i got ur point. when i set "emitter_on_off", for example, in the odd frame, i can collect an image with IR pattern, and in the even frame, without but (in my intuition) illuminated due to "Persistence of vision". then i can just use the even frame as my input, but how can i separate these two kinds of images one by one? there would be any flag in the image stream that i can use to distinguish? i am using "ros2 launch realsense2_camera rs_launch.py" to publish the images via topic. any suggestion will be highly appreciated!!

MartyG-RealSense, 2022-05-20:
A RealSense ROS team member explains at https://github.com/IntelRealSense/realsense-ros/issues/1657#issuecomment-968920694 a method for differentiating between on and off frames in ROS2 (the **ros2** branch) by using the **sequence_id filter** which is part of the High-Dynamic Range (HDR) feature.

MartyG-RealSense, 2022-05-26:
Hi @cheukwaylee  Do you require further assistance with this case, please?  Thanks!

cheukwaylee, 2022-05-26:
Hi. no for now, i close this issue. and if there is anything else that needs your help later, i will reopen it, THANKS!

MartyG-RealSense, 2022-05-26:
You are very welcome.  Thanks very much for the update!

cheukwaylee, 2022-06-01:
Hi! I really would like to know more info about "the position of IR white dot pattern". as u told me before i can only get the pattern as following picture shows, but i donot know the exact position and size of each white dot. and what is the meaning of "91x65deg" in d435?

what i expect to know is,  when the resolution is set as 640x480 for example, the pixel coordinates of each dot, and their radius, and i guess it should be proportional when using other resolutions. THANKS in advance!

[image]

MartyG-RealSense, 2022-06-01:
The projected pattern is semi-random, so you would likely not be able to obtain the position of each dot unless you identify pixels on the infrared image that have a particular **intensity** range.  

Intel have published a white-paper guide to projectors at the link below.

https://dev.intelrealsense.com/docs/projectors

The angle values for each camera model refer to the angle of emission of the projector.  The D435 has a wider emission angle but fewer dots as a consequence.  The dots act as a texture source for the camera to analyze for depth information when they are projected onto a surface / object, so the more dots that there are in a scene the better.  For example, two cameras with overlapping fields of view should provide better depth information due to there being redundancy in the data in the area where the two fields of view and pattern projections overlap.

The white-paper guide has a section dedicated to the dot pattern at the link below that discusses dot pattern density versus resolution.

https://dev.intelrealsense.com/docs/projectors#section-5-the-dot-pattern

cheukwaylee, 2022-06-04:
THANKS for ur reply. I would try with ur suggestion.

and there is another question i would like to ask u.
how can I align the depth image to the left or right image, I can only find the solution to align depth to RGB.

MartyG-RealSense, 2022-06-04:
The depth image is already perfectly aligned to the left IR image as they originate from the same sensor and so depth-IR alignment does not need to be performed.  The SDK documentation provides the following information about this.

*****

The Left IR camera has the benefit of always 1. Being pixel-perfect aligned, calibrated, and overlapped with the depth map, 2. Perfectly time-synchronized, 3. Requires no additional computational overhead to align color-to-depth, and 4. Gives no additional occlusion artifacts, because it is exactly co-aligned.

********

In the RealSense ROS wrapper, only depth to color alignment is supported when align_depth is true.

When aligning to IR in the librealsense SDK instead of the ROS wrapper, a way to align depth, IR and RGB together in the same image is demonstrated in Python code at https://github.com/IntelRealSense/librealsense/issues/5093

closed 2022-06-10 by cheukwaylee