realsenseai/librealsense #10526: Install specific version of librealsense deb package

r91andersson, 2022-05-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.07.150 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    | 4.9.253-tegra                                         | 
| Platform                                 | NVIDIA Jetson NX  |
| SDK Version                            | v2.3.2                          | 
| Language                            | C                          | 
| Segment			| Robot                          | 

### Issue Description

Since there is a problem with librealsense and ros-wrapper on Jetson while using align_depth:=true, I would like to install a specific version of librealsense and ros-wrapper, which I know works.
We previously used librealsense v.2.49.0.0 and ros-wrapper v.2.3.1 with success.

However, when I try to install the librealsense 2.49.0 with the command:
`sudo apt install librealsense2=2.49.0-0~realsense0.5307`

I get the following error:
`E: Version '2.49.0-0~realsense0.5307' for 'librealsense2' was not found`

It was mentioned here on how to install specific versions: https://github.com/IntelRealSense/librealsense/issues/5773#issuecomment-581477785

MartyG-RealSense, 2022-05-18:
Hi @r91andersson   Can you confirm that you have obtained the server key and used the **apt-cache showpkg librealsense2** command to find the correct build number for the 2.49.0 packages, please?

If 5307 is the correct build number, what happens if you use the command below on your Jetson to install the 'utils' and 'dkms' packages required by Intel's official Jetson package instructions?

sudo apt-get install librealsense2-utils=2.49.0-0~realsense0.5307 librealsense2-dkms=2.49.0-0~realsense0.5307

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

r91andersson, 2022-05-18:
@MartyG-RealSense Thanks again!!! Mange to find the correct build tag with the command: **apt-cache showpkg librealsense2**
`sudo apt install librealsense2=2.49.0-5ubuntu5~1.gbpe9f05c librealsense2-dev=2.49.0-5ubuntu5~1.gbpe9f05c`

MartyG-RealSense, 2022-05-18:
Great news!  Thanks very much for the update  :)

r91andersson, 2023-09-12:
@MartyG-RealSense Hi, I just got a new carrier board for our Jetson NX card, and tried the above method to install the same specific version, but now it dosen't seem to provide old releases?

The command `apt-cache showpkg librealsense2 ` only outputs the latest release:

> Package: librealsense2
> Versions: 
> 2.54.1-0~realsense.1425 (/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-arm64_Packages)
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-arm64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-arm64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
> 
> 2.54.1-0~realsense.1423 (/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-arm64_Packages)
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-arm64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_focal_main_binary-arm64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
> 
> 
> Reverse Depends: 
>   librealsense2-dbg,librealsense2 2.54.1-0 ~ realsense.1425
>   librealsense2-dev,librealsense2 2.54.1-0 ~ realsense.1425
>   librealsense2-gl,librealsense2 2.54.1-0 ~ realsense.1425
>   librealsense2-utils,librealsense2 2.54.1-0 ~ realsense.1425
>   librealsense2-utils,librealsense2 2.54.1-0 ~ realsense.1423
>   librealsense2-gl,librealsense2 2.54.1-0 ~ realsense.1423
>   librealsense2-dbg,librealsense2 2.54.1-0 ~ realsense.1423
>   librealsense2-dev,librealsense2 2.54.1-0 ~ realsense.1423
> Dependencies: 
> 2.54.1-0 ~ realsense.1425 - libusb-1.0-0 (2 2:1.0.16) libudev1 (2 183) libc6 (2 2.29) libgcc-s1 (2 3.0) libstdc++6 (2 9) librealsense2-udev-rules (0 (null)) 
> 2.54.1-0 ~ realsense.1423 - libusb-1.0-0 (2 2:1.0.16) libudev1 (2 183) libc6 (2 2.29) libgcc-s1 (2 3.0) libstdc++6 (2 9) librealsense2-udev-rules (0 (null)) 
> Provides: 
> 2.54.1-0 ~ realsense.1425 - 
> 2.54.1-0 ~ realsense.1423 -

Do you know how we can install the same specific version 2.49.0 as we used before?

MartyG-RealSense, 2023-09-12:
Hi @r91andersson  I will check with my Intel RealSense colleagues about this issue.  Thanks very much for your patience.

Nir-Az, 2023-09-12:
Hi @r91andersson ,
Please check if this instructions works for you:
If yes, we will need to update the Jetson instructions

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

r91andersson, 2023-09-13:
@Nir-Az, I tried according to the instructions, but it still only 3 versions available:

> 2.54.1-0 ~ realsense.9591 - 
> 2.54.1-0 ~ realsense.9588 - 
> 2.53.1-0 ~ realsense0.8251 -

We would need version 2.49 (if it possible to release in the apt package), as it is a bug in the later versions when using align_depth:=true. The color stream will, when using align_depth:=true, add a 200ms delay, ending up to 250ms in total. But when using version 2.49 we can use align_depth:=true and have around 30-40 ms total delay of color stream.

Nir-Az, 2023-09-13:
Hi,
Please make sure you run apt-get update 
What I see when I run your command is:
[image]

r91andersson, 2023-09-13:
@Nir-Az I tried run it on the Xavier NX, which by the way has Ubuntu 20.04 (Focal) installed, but I only see the latest release after running the commands + apt-get update. 

Could it be that the older version of realsense sdk is not released to the Focal dist?

Nir-Az, 2023-09-13:
That is an important detail.

And this is correct, JP5 (Based on U20) first official release release was 2.54.1.
Could be that 2.53.1 you see is some version we did to test the Debians integration.

I think this explains it.

r91andersson, 2023-09-13:
Alright, then I understand! So that gives me the only option to build the version from source?

Nir-Az, 2023-09-13:
I think the only solution for you can be running a JP4 docker container on your Jetson NX JP5.

r91andersson, 2023-09-13:
Ok. And the root cause for us to do this is because of the extra delay that is added in the color stream while having the option align_depth:=true.
If we're unable to see this extra delay by using realsense-viewer with the same options, it means that the error is most likely introduced in the ros-wrapper?

MartyG-RealSense, 2023-09-27:
Hi @Nir-Az  Could you help @r91andersson with their question in the comment above, please?  Thanks!

Nir-Az, 2023-09-27:
I am not sure I understand the issue.
1 thing to know is that RS-Viewer use some enhancements that not sure the ROS wrapper use.
CUDA, Open GL...

Please elaborate on the issue and we will try to help.

MartyG-RealSense, 2023-10-03:
Hi @r91andersson  Could you provide further information for @Nir-Az regarding their response in the comment above, please?  Thanks!

MartyG-RealSense, 2023-10-17:
Hi @r91andersson  Do you require further assistance with this case please?  If you do then please provide further information about your issue, as requested by @Nir-Az  above at https://github.com/IntelRealSense/librealsense/issues/10526#issuecomment-1738026844

MartyG-RealSense, 2023-10-23:
Case closed due to no further comments received.

closed 2023-10-23 by MartyG-RealSense

youngfriday, 2024-08-26:
> I think the only solution for you can be running a JP4 docker container on your Jetson NX JP5.
I have tried this way on my Jetson Orin NX (JetPack 5.1),and it turns out to be failed . I build a environment with Ubunto18.04&Jetson 4.6. But it still cannot install the previous version .Docker cannot change the real kernel version of my Jetson, which is a 5.10 for Ubunto20.04. It still doesn't fit for the previous version before 2.50.

youngfriday, 2024-08-26:
@r91andersson Hello ! Have you find a way to fix this problom? Tell me if you have succeeded !
Thank you very much!!!