realsenseai/librealsense #10527: Get transform matrix from camera ro world coordinates

Yiiii19, 2022-05-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455&L515 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { v2.3.2 }                          | 
| Language                            |  {python }                          | 
| Segment			|  { }                          | 

### Issue Description
Hi,
I am quite new to this camera and I am having two cameras D455&L515 for doing the 3D reconstruction.

I am using the algorithm called NeRF which requires the transform matrix, how can I get this one?

I have looked through quite many issues here and I saw we can get extrinsic matrix from sdk which is about aligning color to depth or vice versa. Is this matrix the "transform matrix camera to world"?

Thanks in advance.

MartyG-RealSense, 2022-05-18:
Hi @Yiiii19  If you are aligning color to depth or depth to color on the same time then you should not need to get involved with extrinsics, as the RealSense SDK's **align_to** instruction can perform the calculations for you via its Align Processing Block.  An SDK example program for depth to color alignment in Python at the link below that is compatible with both the D455 and L515 camera models is **align-depth2color.py**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

For color to depth alignment, change line 60 from 'align_to = rs.stream.color' to `align_to = rs.stream.depth`

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py#L60

*********

In regard to obtaining the extrinsic transform between a pair of sensors, you can do this in Python with the **get_extrinsics_to** instruction.  A RealSense team member provides a Python example code for depth and color at https://github.com/IntelRealSense/librealsense/issues/2124#issuecomment-407187377

Alternatively, you can obtain extrinsics for all supported sensor combinations including depth to color and color to depth by using the SDK's rs-enumerate-devices tool, launching it in calibration information mode with the command **rs-enumerate-devices -c**

Yiiii19, 2022-05-18:
Hi,
thanks for your so fast reply.

1. Firstly for your question at the end: I just use one camera either D455 or L515. So far i donot need any transform between thee two cameras.

2. One question about what you replied to me, want to make sure if I understand you correctly: 
(we only talk about depth to color now)

- Through **align-depth2color.py** the results are color and aligned depth imagines
- **get_extrinsics_to** can give me the  extrinsic transform matrix which can **align depth to color**

I just added more description of my question and you can check that again please.

Thanks again!

MartyG-RealSense, 2022-05-18:
The **align_depth2color.py** example program results in depth aligned to color, yes.

If using the **align_to** SDK instruction then you do not need to retrieve the extrinsics in order to perform alignment, as the SDK automatically performs the calculations and adjustments between depth and color for you.

Yiiii19, 2022-05-18:
Great thanks.

Can you or your collegue also answer me the question at the beginning?
Thanks!

> Firstly I am sure the format matrix I want to get is combined of rotation and translation. Or we call it camera pose
I am doing the indoor video streaming, so the camera was kept moving and imagines were captured. The camera coordinate is moving as well.
Hence, I need captured imagines and corresponding matrix from camera to world coordinate for each imagine
The world origin does not matter where it is. But i really donot know how to get this Matrix.

MartyG-RealSense, 2022-05-18:
The pose of the camera is only supported as a built-in feature on the RealSense T265 Tracking Camera model, which supports a **pose stream**.  It is possible to pair a T265 with another RealSense camera model (for example, on the same mounting bracket) so that the T265 can provide pose whilst the other camera provides depth information.  Intel have published a guide about this at the link below.

https://www.intelrealsense.com/depth-and-tracking-combined-get-started/

If your project cannot use a T265 then it should be possible to obtain the camera's relative position on a map using ROS and SLAM.  A guide at the link below that was created by a RealSense ROS user demonstrates a method for doing this.

https://shinkansan.github.io/2019-UGRP-DPoom/SLAM

Yiiii19, 2022-05-18:
T265 camera is possible for me to use. I just look through the example code. I do see the code dealing with **pose stream** but I am not sure which one is correct?

`print("Position: {}".format(data.translation))`
what does the position look like (I cannot test it yet cause i didnot have the T265)?
Thanks!

MartyG-RealSense, 2022-05-19:
It looks as though the quoted line comes from the program **t265_example.py**.  If it does, the image below shows an output from that program.

[image]

Yiiii19, 2022-05-19:
Yes, it comes from **t265_example.py**

1. I do think this position is in camera coordinate. But what does the position mean? like of one point?
2. How can I get the position in world coordinate/camera pose in world coordinate? That is exactly what I really want to know

MartyG-RealSense, 2022-05-19:
The T265's position is in world coordinates, as described in the T265 documentation link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md#sensor-origin-and-coordinate-system

I believe that when the T265 starts up, its position will be 0,0,0 and then as the T265 is moved, its position updates to reflect how many meters it has travelled in world distance from its 0,0,0 start position.

Yiiii19, 2022-05-20:
Thanks for your reply. Quite clear about the T265 camera.

> If your project cannot use a T265 then it should be possible to obtain the camera's relative position on a map using ROS and SLAM. A guide at the link below that was created by a RealSense ROS user demonstrates a method for doing this.
https://shinkansan.github.io/2019-UGRP-DPoom/SLAM

I just have a look through yesterday about that repo, I unfortunately find the method to get the mapping matrix I want. Do you have any idea?

MartyG-RealSense, 2022-05-20:
Regarding obtaining the pose of a camera without a T265, the available possibilities are limited.

Apparently in ROS, RViz has a **2D Pose Estimate** button when RViz has been set up with the **Navigation Stack**.  This enabkes the pose of a robot to be set.  There is a tutorial about this at the link below.

http://wiki.ros.org/navigation/Tutorials/Using%20rviz%20with%20the%20Navigation%20Stack

Yiiii19, 2022-05-23:
Many thanks  for your help!

closed 2022-05-23 by Yiiii19