realsenseai/librealsense #10531: in UE 4 ... ARealSenseInspector::Start exception: Profile not supported

gschian0, 2022-05-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (05.12.14.50) | 
| Operating System & Version |   {Win 10)                                         | 
| Platform                                 | UE 4 |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {UE C++ }                          | 
| Segment			|  {Entertainment }                          | 

### Issue Description

I have created the code files, added the realsense library to the plugins folder... on first open I was able to view the camera.  After connecting and disconnectijng another camera... I started running into an issue ...

`LogRealSense: Display: ARealSenseInspector::Start 000001B4304D5600
LogRealSense: Display: enable_device: SN=102122061338
LogRealSense: Display: enable_stream DEPTH 1280x720 @90
LogRealSense: Error: ARealSenseInspector::Start exception: Profile not supported
LogRealSense: Display: Flush rendering commands
LogRealSense: Display: Destroy dynamic textures`

any help would be appreciated... I have 3 cameras I would like to enumerate and use textures from... I feel like I'm getting close to my goal.

Thanks!
-Gennaro

MartyG-RealSense, 2022-05-19:
Hi @gschian0  90 FPS at 1280x480 depth resolution is not supported on RealSense 400 Series cameras.  The maximum FPS on that resolution is 30.  However, 848x480 depth resolution supports 90 FPS.

gschian0, 2022-05-19:
is there a simple way to change FPS in a setting?  can you point me to a blueprint?

gschian0, 2022-05-19:
I selected the BP_RealSenseInspector, navigated to ... rate on all of the feeds and I am better.  Thanks for the speedy response!

MartyG-RealSense, 2022-05-19:
You are very welcome.  It's excellent to hear that you achieved a solution.  Thanks for the update!

closed 2022-05-23 by gschian0