realsenseai/librealsense #10532: ub Y

gschian0, 2022-05-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415  } | 
| Firmware Version                   | (05.12.14.50) | 
| Operating System & Version |   {Win 10)  | 
| Kernel Version (Linux Only)    |                 | 
| Platform                                 | unreal engine  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C++ unreal engine }                          | 
| Segment			|  {Entertainment }                          

attempting to upgrade project files to ue5... I get this error when trying to generate a new project.

`Running C:/Program Files/Epic Games/UE_5.0/Engine/Binaries/DotNET/UnrealBuildTool/UnrealBuildTool.exe  -projectfiles -project="C:/Users/GennaroSchiano/Documents/GitHub/librealsenseSR/wrappers/unrealengine5/RealSenseUE.uproject" -game -rocket -progress -log="C:\Users\GennaroSchiano\Documents\GitHub\librealsenseSR\wrappers\unrealengine5/Saved/Logs/UnrealVersionSelector-2022.05.19-19.14.36.log"
Log file: C:\Users\GennaroSchiano\Documents\GitHub\librealsenseSR\wrappers\unrealengine5\Saved\Logs\UnrealVersionSelector-2022.05.19-19.14.36.log
Log file: C:\Users\GennaroSchiano\AppData\Local\UnrealBuildTool\Log_GPF.txt

Some Platforms were skipped due to invalid SDK setup: IOS, Android, Linux, LinuxArm64.
See the log file for detailed information

Discovering modules, targets and source code for project...
Compiled assembly file 'C:\Users\GennaroSchiano\Documents\GitHub\librealsenseSR\wrappers\unrealengine5\Intermediate\Build\BuildRules\RealSenseUEModuleRules.dll' appears to be for a newer CLR version or is otherwise invalid.  Unreal Build Tool will try to recompile this assembly now.  (Exception: Could not load file or assembly 'C:\Users\GennaroSchiano\Documents\GitHub\librealsenseSR\wrappers\unrealengine5\Intermediate\Build\BuildRules\RealSenseUEModuleRules.dll'. Format of the executable (.exe) or library (.dll) is invalid.)
C:\Users\GennaroSchiano\Documents\GitHub\librealsenseSR\wrappers\unrealengine5\Plugins\RuntimeMeshComponent\Source\RuntimeMeshComponent\RuntimeMeshComponent.Build.cs(11,9): error CS0103: The name 'bFasterWithoutUnity' does not exist in the current context
C:\Users\GennaroSchiano\Documents\GitHub\librealsenseSR\wrappers\unrealengine5\Plugins\RealSense\Source\RealSense\RealSense.Build.cs(55,4): error CS0103: The name 'PublicLibraryPaths' does not exist in the current context
C:\Users\GennaroSchiano\Documents\GitHub\librealsenseSR\wrappers\unrealengine5\Plugins\RealSense\Source\RealSense\RealSense.Build.cs(56,4): error CS0103: The name 'PublicLibraryPaths' does not exist in the current context
ERROR: Expecting to find a type to be declared in a target rules named 'RealSenseUETarget'.  This type must derive from the 'TargetRules' type defined by Unreal Build Tool.`

thanks for any help in advance! 
-Gennaro

MartyG-RealSense, 2022-05-20:
Hi @gschian0  The RealSense Unreal wrapper is designed for UE 4.26 and has references in its code that perform version checks for that version number, as shown at https://github.com/IntelRealSense/librealsense/commit/20a7930681eed73ddd2475646a29bed0abf02d59#diff-6ef71b6febaaf99bfc1a6e6690679392d8afd0bc8eac13a56283990cb37fd7d0R1

Particularly in the wrapper file **RealSenseUE.uproject**:

`"EngineAssociation": "4.26",`

DryreL, 2023-12-07:
I have same problem on UE 5.3

[image]

MartyG-RealSense, 2023-12-07:
Hi @DryreL  The RealSense UE4 wrapper will not work with UE5.  A RealSense user shared a RealSense plugin for UE5 at https://github.com/IntelRealSense/librealsense/issues/12262