realsenseai/librealsense #10533: How to apply the Post processing filter ?

Amayess, 2022-05-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455  | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   {Win10  |                                  | 
| Platform                                 | PC |
| SDK Version                            | 2.50.0 }                          | 
| Language                            |  {C++}                          | 
| Segment			|  {Robot}                          | 

### Issue Description
Hi,
Am actually creating an app using c++ and Qt. So far i have been able to link my C++ code with Qt and display the intel realsense frames on it with no problem.

Actually, I am trying to apply the post processing filters like in the SDK example but on my own created app but i see no results ? still too much temporal noise and no smooth frames :/ 

This is the code i used for : 

`frames = pipe.wait_for_frames().
                 apply_filter(dec_filter).
                 apply_filter(thr_filter).
                 apply_filter(spat_filter).
                 apply_filter(temp_filter);
                 auto color = frames.get_color_frame();`
                  auto depth = frames.get_depth_frame();
    emit newImage(frameToQImage(frames.get_depth_frame().apply_filter(color_map).apply_filter(dec_filter).apply_filter(temp_filter)));

Is my approach correct ? 

Can someone help me to solve it please ?

MartyG-RealSense, 2022-05-20:
Hi @Amayess  I have looked carefully at the provided information but it is difficult to provide advice based on the partial code above.  

The general structure of a C++ post-processing script would be to (1) define the filters before the wait_for_frames() instruction, and (2) apply the defined filters with a **process** instruction after the wait_for_frames instruction, like with the C++ example in the SDK's post-processing documentation.

https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#using-filters-in-application-code

```
// Streaming initialization
    rs2::pipeline pipe;

    // Declare filters
    rs2::decimation_filter dec_filter;
    rs2::spatial_filter spat_filter;

    // Configure filter parameters
    dec_filter.set_option(RS2_OPTION_FILTER_MAGNITUDE, 3);

    spat_filter.set_option(RS2_OPTION_FILTER_SMOOTH_ALPHA, 0.55f);

    // Main Loop
    while (true) {
       rs2::frameset data = pipe.wait_for_frames();
       rs2::frame depth_frame = data.get_depth_frame();

       rs2::frame filtered = depth_frame;
       // Note the concatenation of output/input frame to build up a chain
       filtered = dec_filter.process(filtered);
       filtered = spatial_filter.process(filtered);

    }
```

Amayess, 2022-05-20:
Hi @MartyG-RealSense 
Thank you for you answer, actually this is exactly what i added to my code but still can't see no difference :/ 

it seems like no updated is done to my frames and depth camera after the post processing
[image]

Amayess, 2022-05-20:
Hi @MartyG-RealSense
Thank you for you answer, actually this is exactly what i added to my code but still can't see no difference :/

it seems like no updated is done to my frames and depth camera after the post processin
[image]
g

MartyG-RealSense, 2022-05-20:
In your **emit newImage** line, as you have already defined **auto depth = frames.get_depth_frame()** earlier in the script I would think that 'depth' would be simply put into the newImage instruction instead of repeating the frames.get_depth_frame instruction again.

`emit newImage(frameToQImage(depth.apply_filter(color_map).apply_filter(dec_filter).apply_filter(temp_filter)));`

MartyG-RealSense, 2022-05-20:
The default magnitude value of the decimation filter is '2', meaning patches that are sized 2x2 pixels.  The post-processing filter states the the supported values for the magnitude are 2 to 8, so setting it to '1' should cause a failure.  

https://dev.intelrealsense.com/docs/post-processing-filters#section-decimation-filter

If you do not specify a magnitude then it should use '2' by default.  It cannot be set lower than that.

MartyG-RealSense, 2022-05-27:
Hi @Amayess  Do you require further assistance with this case, please?  Thanks!

Amayess, 2022-05-30:
Hi @MartyG-RealSense, thank you for your feedback 

I have one last question, what are the best parameters for the filters and what filter should i use ? so i can have an optimized stream to generate a clean the mesh.

Thank you.

MartyG-RealSense, 2022-05-30:
For the decimation filter, the selected resolution will be reduced according to the magnitude value.  So if 848x480 is used then a magnitude of 2' will reduce the resolution to 424x240, whilst a magnitude of '3' would give a resolution of about 284x160.  I would recommend trying the default of '2' or the '3' that you have used already in your script.  I have very rarely seen larger values than '3' used, even though the decimation scale goes up to '8'.

The default value for the Spatial filter is '2'.

Amayess, 2022-05-30:
Thank you @MartyG-RealSense :) !

Amayess, 2022-06-05:
Hi @MartyG-RealSense , no thank you for you help, you can close it.

MartyG-RealSense, 2022-06-05:
Thanks very much for the update!  As you suggest, I will close the issue.  Thanks again!

closed 2022-06-05 by MartyG-RealSense