realsenseai/librealsense #10535: realsense viewer does not detect device

Shawn207, 2022-05-20:
----------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i |  
| Operating System & Version |   Ubuntu20 (WSL2 on win 10)  | 
| Kernel Version (Linux Only)    |  5.10                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2                        | 
                          

### Issue Description
I followed the instruction under doc: linux_distribution, but the realsense-viewer does not detected the device. Then I noticed this method may no support kernel 5.10, so I tried the libusb_installation. The installation is completed but the problem still holds. Could you please give me any suggestions?

Shawn207, 2022-05-20:
Yes, it was libuvc backend installation method, so I agree that it's not the problem of kernel.  The camera was detected by windows because both Depth and RGB are shown in Device Manager/Camera. I also checked the privacy settings and the settings should be fine( camera access for this device is on and allow apps to access your camera is on)

Shawn207, 2022-05-20:
One more thing is I justed downloaded realsense-viewer on windows following [https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_windows.md](url), and it works for windows. I don't know why it fails on ubuntu

MartyG-RealSense, 2022-05-20:
I will continue researching this case and your related one at https://github.com/IntelRealSense/realsense-ros/issues/2356 tomorrow.  References for using WSL2 with RealSense cameras are rare though and have typically ended in failure in the past, except for the recent success reported by the RealSense user at https://github.com/IntelRealSense/librealsense/issues/10506#issuecomment-1125860744 after Microsoft's USB update.

Thanks very much for your patience, and please do post updates on your cases if you make progress in the meantime.  Good luck!

By the way, you could also use the **rs-enumerate-devices** tool and the Linux **lsusb** terminal command to check whether the camera is detectable even if it is not appearing in realsense-viewer on Ubuntu.

closed 2022-06-02 by MartyG-RealSense