realsenseai/librealsense #10536: The orientation of exported PLY file from realsense-viewer

gywhitel, 2022-05-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |    Linux (Ubuntu 18)   | 
| Platform                                 | PC|

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello, I am using realsense-viewer to export PLY files for further processing. Yet I am confused of the coordinates of the exported PLY files. The figure below shows two coordinate systems. The left one shows the PLY file, and the right one is from realsense-viewer. Their Y axes and Z axes are antiparallel.

[image]

For example, I pick a point (x, y, z) in realsense-viewer, yet the point is converted to (x, -y, -z) in PLY.
So if I got a point (a,b,c) from PLY, I should convert it to (a,-b,-c)  to coordinate with actual realsense camera frame ?

MartyG-RealSense, 2022-05-21:
Hi @gywhitel  The reason for the flipping of the axes of a ply exported from the RealSense Viewer is explained at https://github.com/IntelRealSense/librealsense/issues/3589

According to that case, this flipping of axes was implemented in SDK version 2.19.0.  If you would prefer the axes to be unflipped, you could try using version **2.18.1** of the Viewer.  You can download this by visiting the link below and left-clicking on the **Intel.RealSense.Viewer.exe** file in the 'Assets' file list at the bottom of the page.  Once downloaded, launch the Intel.RealSense.Viewer.exe file from the folder location on your computer that you downloaded it to.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.18.1

[image]

If you would prefer a modern SDK with unflipped axes, conceivably you could build a customized version of the SDK from source code with the changes at https://github.com/IntelRealSense/librealsense/commit/9e4b6ba09ac6fd0b20bc28f388e44791f4cc04c1 that were made to the SDK to implement axis flipping reversed.

gywhitel, 2022-05-22:
Appreciate your reply. But what I really want to figure out is how to transform the point coordinates in exported PLY files to realsense camera coordinates, instead of changing the SDK.
Please check the following illustration:
> **So if I got a point (a,b,c) from PLY, I should convert it to (a,-b,-c) to coordinate with actual realsense camera frame ?**

Is my understanding right ?

MartyG-RealSense, 2022-05-22:
That is my interpretation of the conversion line in the SDK code, yes - that it is preserving the original value of X and flipping Y and Z to minus sign.

`new_vertices.push_back({ vertices[i].x,  -1*vertices[i].y, -1*vertices[i].z });`

gywhitel, 2022-05-22:
Thanks for your reply. I'll manually convert the coordinates into camera frame.

closed 2022-05-22 by gywhitel