realsenseai/librealsense #10537: Programs can't detect T265 but viewer can in Jetson nano

RonghaiHe, 2022-05-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.951 | 
| Operating System & Version |   Ubuntu18.04  | 
| Kernel Version (Linux Only)    |  4.9.253                                        | 
| Platform                                 | NVIDIA Jetson Nano  |
| SDK Version                            |  2.50.0                          | 
| Language                            |  python &C++                         | 
| Segment			|  Robot                          | 

### Issue Description
I connected T265 to Jetson Nano and I can start obtaining the tracking information in the **realsense-viewer** with Usb Type Descriptor: **USB3.1**. When I **closed the viewer** and tried running the python codes **t265_samples.py** in /librealsense/wrappers/python/examples, it showed that **no devices connected**.
```
Traceback (most recent call last):
  File "t265_example.py", line 21, in <module>
    pipe.start(cfg)
RuntimeError: No device connected
```
When I tried running the **rs-pose**, a compiled C++ program in /librealsense/build/examples/pose, it showed that: 
```
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x559cc0d7b0):
    Frame didn't arrive within 15000
```
I entered `rs-enumerate-devices -d` and it showed the Device info as well as (outputs about the time are deleted):
```
Default streams:
(handle-libusb.h:100) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
(librealsense-exception.h:52) Unable to open device interface
(handle-libusb.h:100) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
(librealsense-exception.h:52) Unable to open device interface
(handle-libusb.h:100) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
(librealsense-exception.h:52) Unable to open device interface
RealSense error calling rs2_config_resolve(config:0x558a34ec00, pipe:0x558a34ccb0):
```
When I connected **D435i** with Jetson Nano in the same USB interface and run C++ programs and python codes, there's **nothing wrong** with them.

The **installation steps** are as below. I don't know whether it made a difference to this issue: 
At first, I use this way to install the RealSense SDK: 
[Running pyrealsense2 on JetsonNano](https://github.com/IntelRealSense/librealsense/issues/6964)
I also copied .so documents from /librealsense/build/wrappers/python to /usr/lib/pythn3/dist-packages
Then I noticed [librealsense/doc/installation_jetson.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetsonRunning%20pyrealsense2%20on%20JetsonNano.md) and I just did "**Getting started**" because I thought that's enough.

In conclusion, I can obtain the information from T265 in the realsense-viewer but I can't do it by python codes or C++ programs.

NemesisX09, 2022-11-27:
I am running into the same issue, was anything else done other than waiting? Does it cease to work every so often? Thanks!

jyomk, 2023-08-01:
I'm trying to run the T265 camera for a tracking project. The camera is detected using lsusb, rs-enumerate-devices. Also, when running rs-pose, I see the coordinates. however, when I start a pipeline I get the following error message: RealSense error calling rs2_pipeline_start_with_config(pipe:0x55f0ab2e0e50, config:0x55f0ab2e1400) Has anyone faced this issue? Please let me know if you have any solution. Thanks!

closed 2026-10-23 by MartyG-RealSense