realsenseai/librealsense #10538: How to use a frame queue on a recorded bag file to prevent dropping any frames

JeffR1992, 2022-05-22:
### Issue Description
Your frame queue example (https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/frame_queue_example.py) doesn't really tell us how to actually use frame queues, and it would be nice if it was expanded upon. For example, I'm trying the following:

```
file_path = "path/to/bag/file"
queue_size = 100

config = rs.config()
config.enable_device_from_file(file_path, repeat_playback=False)

pipeline = rs.pipeline()
pipeline.start(config)

while True:
    frames = pipeline.wait_for_frames()

    infrared_frame = frames.get_infrared_frame()
    infrared_image_array = np.asanyarray(infrared_frame.get_data())

    depth_frame = frames.get_depth_frame()
    depth_image_array = np.asanyarray(depth_frame.get_data())
```

If I now replace a few lines to try and use a frame queue like the following:

```
file_path = "path/to/bag/file"
queue_size = 100

config = rs.config()
config.enable_device_from_file(file_path, repeat_playback=False)

frame_queue = rs.frame_queue(queue_size, keep_frames=True)

pipeline = rs.pipeline()
pipeline.start(config, frame_queue)

while True:
    frames = frame_queue.wait_for_frame()

    infrared_frame = frames.get_infrared_frame()
    infrared_image_array = np.asanyarray(infrared_frame.get_data())

    depth_frame = frames.get_depth_frame()
    depth_image_array = np.asanyarray(depth_frame.get_data())
```

I keep getting the following error:

```
Traceback (most recent call last):
  File "c:\Users\jeffr\Desktop\Code\test.py", line 21, in <module>
    infrared_frame = frames.get_infrared_frame()
AttributeError: 'pyrealsense2.pyrealsense2.frame' object has no attribute 'get_infrared_frame'
```

So my question is, how do I actually use a frame queue to play back an infrared, color and/or depth stream, from a recorded bag file, without dropping any frames?

MartyG-RealSense, 2022-05-23:
Hi @JeffR1992  I note that your script does not seem to define any stream types in the config section of your script.  Even if you are reading from a bag file instead of a live camera, the streams still need to be defined.  This principle is demonstrated in the SDK's Python **read_bag_example.py** sample program.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py#L46

For example, to define depth, infrared and color streams:

```
config = rs.config()
config.enable_stream(rs.stream.depth, rs.format.z16, 30)
config.enable_stream(rs.stream.infrared, 640, 480, rs.format.y8, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)
```

It is also worth highlighting that the frame queue size can be defined with the **rs.option_frames_queue_size** instruction, as demonstrated at https://github.com/IntelRealSense/librealsense/issues/6448#issuecomment-633041074

MartyG-RealSense, 2022-05-29:
Hi @JeffR1992  Do you require further assistance with this case, please?  Thanks!

rubenjanss, 2022-06-01:
Hi @MartyG-RealSense , I am not the OP, but am running into the same problem. Even when defining the stream type, I get the error `AttributeError: 'pyrealsense2.pyrealsense2.frame' object has no attribute 'get_color_frame'`.

This is the current code:

`file_path = "path/to/file.bag"

pipeline = rs.pipeline()
config = rs.config()

rs.config.enable_device_from_file(config, file_path)
config.enable_stream(rs.stream.color, rs.format.rgb8, 30)

queue = rs.frame_queue(50, keep_frames=True)

pipeline.start(config, queue)

while True:
    frames = queue.wait_for_frame()

    color_frame = frames.get_color_frame()

    color_image = np.asanyarray(frames.get_data())

pipeline.stop()`

Could you provide an example usage of frames coming from a queue?

MartyG-RealSense, 2022-06-02:
Hi @rubenjanss  Is the advice in https://github.com/IntelRealSense/librealsense/issues/6271 helpful to your problem?  https://github.com/IntelRealSense/librealsense/issues/6271

MartyG-RealSense, 2022-06-08:
Hi @JeffR1992 and @rubenjanss  Do you require further assistance with this case, please? Thanks!

closed 2022-06-14 by MartyG-RealSense