realsenseai/librealsense #10539: D455 Question regarding depth frame

MartinPedersenpp, 2022-05-23:
Hello all

I am trying to do some projection from 2D to 3D using my Realsense D455.
I have an issue where objects of different heights are getting slightly skewed in the 3D space based on their height.
Example:
I place a very short object in front of my camera in the center pixel 640x360 (center of 1280x720 resolution). In 3D space this gets placed 6cm to the right of (0,0,0). This is expected because the RGB sensor and depth sensor are offset on the camera unit.
I then place a slightly higher object on top of my short object and all of a sudden the 3D capture gets pushed farther away from the 3D center.
If the D455 use stereoscopic capturing to create the depth frame, why is the taller object getting skewed further away when both the left and the right sensor still have the same distance to the object?

Note:
I think by formulating my question I might have answered it as well.
Due to the offset between depth and RGB sensors, the center pixel of the RGB is actually 6cm away from the depth sensors center, so the closer I move an object to the sensors, the closer it will get to either the left or right depth sensor, hence skewing the 3D capture?

Any thoughts or help on this would be greatly appreciated.

Best regards
Martin

MartyG-RealSense, 2022-05-23:
Hi @MartinPedersenpp  I would recommend first exploring the possibility of whether the stacking of the two objects is creating a **repetitive pattern** that is confusing the camera's depth sensing.  A repetitive pattern can be caused by the placement of similar looking objects in a horzontal or vertical pattern (such as a horizontal row of fence posts or tree-tops, a vertical stack of window blind slats or a repeating pattern of floor or ceiling tiles).  

If the cause of the 3D depth distortion is a repetitive patttern then he Intel guides at the links below provide advice about reducing the impact of this phenomenon.

https://dev.intelrealsense.com/docs/depth-map-improvements-for-stereo-based-depth-cameras-on-drones#section-v-incorrect-depth-values

https://dev.intelrealsense.com/docs/mitigate-repetitive-pattern-effect-stereo-depth-cameras

MartinPedersenpp, 2022-05-23:
Update:
So it turns out that there is no distortion in the realsense viewer, but when I capture my depth frame and turn it into a point cloud with Open3D something is messing up, I don't know if you have any experience with the mix of pyrealsense2 and open3D or know why this would be happening, but it seems like the depth map captured by pyrealsense i skewed compared to what happens in the realsense viewer.

closed 2022-05-23 by MartinPedersenpp

MartyG-RealSense, 2022-05-23:
Thanks so much for the update and for sharing the discovered cause of your problem!

MartinPedersenpp, 2022-05-24:
@MartyG-RealSense You are welcome, it would be stupid of me to just leave it open when the issue has been solved.