realsenseai/librealsense #10542: D435 fails to steam on restart

sonali20, 2022-05-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.13.0-41-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.50.0                         | 
| Language                            |  roscpp                         | 
| Segment			|  Robot                         | 

### Issue Description
I have a PC with a D410 and D435 connected to it. Streaming is started with a ROS launch file and is stable the first time. But if I stop and relaunch the streaming, the D410 streams fine but the D435 fails to stream and shows this error -
```
12/05 16:54:47,320 ERROR [140303117178624] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_IO
 12/05 16:54:47,321 ERROR [140303142688512] (sensor.cpp:572) acquire_power failed: failed to set power state
 ```
The camera doesn't even show on lsusb anymore. The only way to recover from this state is a power cycle or an unplug-replug. Relaunching or a soft reboot of the PC doesn't help. I have tried swapping out the camera cable and the result is still the same. This setup doesn't have this problem when both cameras are D415s

MartyG-RealSense, 2022-05-23:
Hi @sonali20   Please try adding the command **initial_reset:=true** to your roslaunch instruction to reset the camera automatically at launch if you are not using this command already.  For example:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

It is also worth highlighting that kernel 5.13 is not supported officially by the librealsense SDK.  The most recent supported kernels are 5.4 if installing librealsense from Debian packages, and 5.8 or 5.11 if building librealsense from source code with CMake.  Whilst unsupported kernels can work with the SDK, there may be unpredictable consequences in regards to stability.

MartyG-RealSense, 2022-05-30:
Hi @sonali20  Do you require further assistance with this case, please?  Thanks!

sonali20, 2022-05-31:
@MartyG-RealSense I was mistaken. The latest firmware was not on the cameras. It was an older version. Once it was upgraded to 5.13.0.50, this problem went away. Thank you for your assistance.

closed 2022-05-31 by sonali20

MartyG-RealSense, 2022-05-31:
You are very welcome, @sonali20 - I'm pleased to hear that you found a solution.  Thanks for the update!