realsenseai/librealsense #10543: D415 got stuck/frooze in wait_for_frame() after unplug-replug

LLY13, 2022-05-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 5.13   just updated today  successfully   | 
| Operating System & Version |    Win 10                                                | 
| Platform                                 | PC  |
| SDK Version                            | updated to latest 2.50.x                           | 
| Language                            |  c++                         | 
| Segment			|  Robot | 

### Issue Description
D415 got stuck  in wait_for_frame() after unplug-replug about 45 seconds occasionally.  I just updated everything today but it still happend in my code.  So I copied demo code, still got stuck. I need know it's normal situation or exception. In the image, I unplugged and replugged immediately.
[image]

MartyG-RealSense, 2022-05-24:
Hi @LLY13  It is possible for a program to become unresponsive after the camera is disconnected by some means (such as physically unplugging it from the USB port).  RealSense users have typically created workaround mechanisms when frames stop arriving, such as automatically resetting the camera with the **hardware_reset()** instruction.  A C++ case that discusses the issue is at https://github.com/IntelRealSense/librealsense/issues/10084

LLY13, 2022-05-25:
I have tried use hardware_reset() before this I posted this issue, seemed like useless.  I'd just try to not unplug it.

MartyG-RealSense, 2022-05-25:
When you unplug, how long on average do you wait before re-inserting?  Upon disconnection, the pipeline should be able to resume streaming if the camera is reconnected within 5 seconds, otherwise the connection time-outs.

LLY13, 2022-05-26:
less than 1 seconds. yep most time reconnected in 5 seconds. but somtimes 15s even 45s+

MartyG-RealSense, 2022-05-26:
I would recommend trying a wait period of 2 seconds before re-inserting to give the computer time to properly register the removal of the device from the USB system and see whether that reduces the occurrences of the long time-out events.

MartyG-RealSense, 2022-06-01:
Hi @LLY13  Do you require further assistance with this case, please?  Thanks!

LLY13, 2022-06-02:
no thx

MartyG-RealSense, 2022-06-02:
Thanks very much @LLY13 for the update.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2022-06-02 by MartyG-RealSense