realsenseai/librealsense #10544: Windows 10 Error opengl3.cpp:572

jondave, 2022-05-24:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:572

```

MartyG-RealSense, 2022-05-24:
Hi @jondave  Which RealSense camera model are you using, please?

Also, if you are using versuion 2.50.0 of the RealSense Viewer then does version **2.49.0** of the standalone .exe version of the Viewer produce the same error?  This can be downloaded by clicking on the **Intel.RealSense.Viewer.exe** link in the Assets list of the release information for 2.49.0 at the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.49.0

[image]

The Viewer .exe should be launched from the folder location that it is downloaded to and not with the launch shortcut icon on the Windows desktop.

jondave, 2022-05-25:
Hi

I am using D435i and T265.
I can open and use the rs-capture in the SDK (Intel.RealSense.SDK-WIN10-2.50.0.3785) tools folder and the cameras work that way.
I have tried both the standalone Intel.RealSense.Viewer.exe and the realsense-viewer.exe in the tools folder and neither will open up. 
Previously when trying to open up this error displayed;
`GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:572`
Now when I try and open the viewer only a small window with the Windows minimise, maximise and close buttons opens up.

Thanks

MartyG-RealSense, 2022-05-25:
Does the Viewer expand to full size if you click on the maximize button on the small window?

jondave, 2022-05-26:
Hi

Launching the 2.49.0 viewer produces a different error,
`The program terminal due to an unrecoverable seh exception access violation`
Following other threads on here I looked in device manager and removed duplicate realsense devices.
But it still 2.49.0 viewer produces the same error.
The 2.50.0 viewer again only show the small window with the min, max and close buttons which automatically closes after a few seconds. 

[image]

MartyG-RealSense, 2022-05-26:
I recalled that this error has occurred in a past case at https://github.com/IntelRealSense/librealsense/issues/9366 with the Viewer specifically.  I suspected that the Viewer's **GLSL** graphics options, which are enabled by default in the Settings interface, were responsible and suggested trying standalone Viewer version **2.21.0**, the version before the GLSL options were introduced.  The RealSense user in that case found that they did not experience the error with 2.21.0.

Could you try downloading and running the standalone 2.21.0 Viewer at the link below to see whether the error still occurs for you, please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.21.0

[image]

jondave, 2022-05-26:
When opening viewer 2.21.0 it still only opened the max, min, close buttons but this time I could expand the window to show the viewer it had detected the realsense camera but no data was displayed.
Closed viewer 2.21.0 and opened viewer 2.49.0 and that works perfectly, opened viewer 2.50.0 and again 2.21.0 and they also work perfectly.
So the problem is fixed by just running viewer 2.21.0 and the other versions work fine.

Thanks for the help and quick responses.

closed 2022-05-26 by jondave