realsenseai/librealsense #10545: Links to 5.13.0.50 realsense firmware link to a PDF and not a firmware

PhilipAtLabrador, 2022-05-24:
I am trying to download firmware 5.13.0.50 to be used on D435. However all of the links I have found point to a PDF and not a firmware.

For example on this page:
https://dev.intelrealsense.com/docs/firmware-releases

The hyperlink for the 5.13.0.50 firmware points to the following link which downloads a PDF document.
https://www.intelrealsense.com/download/19295/?_ga=2.174599650.1924004545.1653433303-1921150582.1653433303

Where can I download this firmware?

MartyG-RealSense, 2022-05-25:
Hi @PhilipAtLabrador  I can confirm that this firmware download link for 5.13.0.50 on the 400 Series firmware releases page leads to a PDF instead of the firmware bin file.  The link below describes an alternative method to update the camera firmware to 5.13.0.50 through the RealSense Viewer, though I appreciate that this will not be suitable in situations where the file itself is required.

https://support.intelrealsense.com/hc/en-us/community/posts/6456192756371

MartyG-RealSense, 2022-05-26:
**Update:** Intel have now corrected the link on the firmware releases page.  Thanks again for your report!

closed 2022-05-26 by PhilipAtLabrador