realsenseai/librealsense #10546: Fail to  var cxt = new Intel.RealSense.Context() in vs2019  c#

sukeyu81, 2022-05-25:
I install thelibrealsense.x64(v2.45.1) and Intel.RealSenseWithNativeDll (v2.47.0) by vs2019 Nuget   in window7

when debug  using AnyCPU  or x64  ,there is something wrong :

when run into  new Intel.RealSense.Context() , or  new pipeline( )
the program exit ,but no anymessage, the return value is  -2147483645 (0x80000003)。

if  debug use x86 ,there is  System.TypeInitializationException:““Intel.RealSense.Context

if anyone know why,thanks

MartyG-RealSense, 2022-05-25:
Hi @sukeyu81  Your problem may be similar to the one described at https://github.com/IntelRealSense/librealsense/issues/4495

It is advised not to use the AnyCPU setting.

There are separate .dll library files in the RealSense SDK for x86 (32-bit) and x64 (64-bit).  So if your project is using x86 then you should use the x86 version of realsense2.dll and if the project is set as x64 then it should use the x64 version of realsense2.dll.  Can you confirm please that you are matching a particular CPU architecture setting with the correct dll file in your project, please?

If you are using Windows and have installed the full RealSense SDK then you can find pre-built x86 and x64 dll files at the following folder location:

**C: > Program Files (x86) > Intel RealSense SDK 2.0 > bin**

[image]

I would also recommend not mixing files from different version numbers together (for example, librealsense 2.45 and Intel.RealSenseWithNativeDll 2.47).  https://github.com/IntelRealSense/librealsense/issues/7307 is an example of the two version numbers being matched.

Intel's official **C# Cookbook** page of C# examples use **var ctx = new Context()** so you could also try that instead of `var ctx = new Intel.RealSense.Context()`

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md

sukeyu81, 2022-05-25:
According  to #7307  ,it worked.
Thank you very much!

MartyG-RealSense, 2022-05-25:
That's excellent news, @sukeyu81 - thanks very much for the update!

closed 2022-05-31 by MartyG-RealSense