realsenseai/librealsense #10548: Getting "404 Stream not found" from code

mhadzic2, 2022-05-26:
Hello,

I have successfully connected my Intel® RealSense™ Camera D435i to Raspberry Pi with rs-server installed on it. It recognizes the camera and shows the server status as running. On my Windows machine, when I use the RealSense Viewer, I am able to connect to the camera using it's IP address and providing the port 8554. 
However, when I try to connect to it using the Verilook SDK by providing the Url, it successfully locates the camera on the network but I keep getting error "404 Stream not found" when it tries to read from the stream.

In RsServer.cpp I noticed there is an url that is generated and is to be used for streaming. The SDK is using RTSP for communication. What is the default URL that can be used for consuming camera's stream?

MartyG-RealSense, 2022-05-26:
Hi @mhadzic2  The RealSense SDK's networking tool does not have a fixed default URL for streaming.  Once the Pi has been logged into via a shell then a **static IP address** should be defined with the procedure in the networking tool's user guide at the section linked to below.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-6-setting-static-ip

Intel recommend using **Putty** to make the ssh connection on Windows, like a RealSense user at https://github.com/IntelRealSense/librealsense/issues/10172 who was also trying to connect from Windows and failing was using.

https://www.putty.org/

mhadzic2, 2022-05-30:
@MartyG-RealSense I know that the URL depends/contains the IP address of the device, but the remainder of the URL is standardized. By reading through the code and comparing to what I see in realsense-viewer, I noticed couple of constants so now when I try to connect to the camera with URL: "rtsp://<<ip_address>>:8554/RGB Camera" it seems to be finding the stream but I am now getting new error: **"Failed to initiate subsession".**

MartyG-RealSense, 2022-05-30:
Your situation sounds similar to a RealSense user in a March 2022 case at https://github.com/IntelRealSense/librealsense/issues/10345#issuecomment-1083005047 who was trying to access the networked RGB stream with **VLC Player** instead of Verilook and could access it with the Viewer but not with VLC.

At a guide that was using **GStreamer** as the video streaming platform at https://ardupilot.org/copter/docs/common-realsense-depth-camera.html they were able to access the video stream with **rtsp://<ip-address>:8554/d4xx** (using d4xx instead of RGB Camera).

You are of course substituting your real IP address in place of <<ip_address>>, yes?

closed 2022-06-12 by MartyG-RealSense