realsenseai/librealsense #10550: Jetson Nano compile c++

remilamoureuxlevesque, 2022-05-27:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 0.5.13.00.50 | 
| Operating System & Version | Linux (Ubuntu 18) | 
| Kernel Version (Linux Only)    | 4.9.253-tegra aarch64                            | 
| Platform                                 | Jetson nano  |
| SDK Version                           |  2.50.0                          | 
| Language                               |  C/C++                         | 
| Segment			         | others                           | 

### Issue Description

I have install the librealsense2 on my jetson nano from the guide on the website (https://dev.intelrealsense.com/docs/nvidia-jetson-tx2-installation) with the Install with debian packages method. After that I was able to run the realsense-viewer and some example (rs-hello- realsense). Then I try to copy the code from the realsense example and compile it with the line provide at the end of the installation guide. 

```g++ -std=c++11 realsense-hello.cpp -lrealsense2```

But when I do this I got the following error.
```
realsense-hello.cpp:7:1: error: ‘p’ does not name a type
 p.start();
 ^
realsense-hello.cpp:23:6: error: ‘cout’ in namespace ‘std’ does not name a type
 std::cout << "The camera is facing an object " << dist_to_center << " meters away \r";
      ^~~~
```
I would like to make it work before starting developping my own code.

This is the code I try to compile : 

realsense-hello.cpp
```
#include <librealsense2/rs.hpp> // Include Intel RealSense Cross Platform API

// Create a Pipeline - this serves as a top-level API for streaming and processing frames
rs2::pipeline p;

// Configure and start the pipeline
p.start();

// Block program until frames arrive
rs2::frameset frames = p.wait_for_frames();

// Try to get a frame of a depth image
rs2::depth_frame depth = frames.get_depth_frame();

// Get the depth frame's dimensions
float width = depth.get_width();
float height = depth.get_height();

// Query the distance from the camera to the object in the center of the image
float dist_to_center = depth.get_distance(width / 2, height / 2);

// Print the distance
std::cout << "The camera is facing an object " << dist_to_center << " meters away \r";
```
How can I make it work ? Let me know if I should update something or add anything,

Thanks in advance for your help.

MartyG-RealSense, 2022-05-27:
Hi @remilamoureuxlevesque  The SDK example programs use a different build method from the typical approach that would be used to create your own program, so they are not a good tutorial about how to set up an independent project and create the linkages to the librealsense library files.  This subject is discussed regarding rs-hello-realsense at https://github.com/IntelRealSense/librealsense/issues/6797

Intel provide a simple **Getting Started** project template at the link below.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

**rs-getting-started.cpp** is essentially the same script as rs-hello-realsense.cpp.  The main difference seems to be in its CMakeLists.txt build file, which is more complex than the one that rs-hello-realsense uses and includes additional instructions for finding the librealsense library.

**CMakeLists.txt - rs-hello-realsense**
https://github.com/IntelRealSense/librealsense/blob/master/examples/hello-realsense/CMakeLists.txt

**CmakeLists.txt - rs-getting-started**
https://github.com/zivsha/librealsense/blob/getting_started_example/examples/getting-started/CMakeLists.txt#L20-L31

Intel also provide an introductory level CMake example project for Linux that demonstrates simple linking to the librealsense library.

https://github.com/IntelRealSense/librealsense/tree/master/examples/cmake

The concept explained in that project of using the **target_link_libraries** instruction to create linkages to librealsense is also discussed at https://github.com/IntelRealSense/librealsense/issues/4277#issuecomment-505242118

remilamoureuxlevesque, 2022-05-27:
Thanks for your help, after doing the getting started procedure, it works. 

I have one more question, after cloning the getting-started-examples branch I look for the /examples/getting-started folder and I can't find it. Is it normal ? I must copy every file by hand to make it work.

Thanks again for your help.

MartyG-RealSense, 2022-05-27:
I suspect that the default 'git clone' command is cloning the **master** branch that does not have the getting-started-examples folder, and you need to configure the git clone command to configure a **specific** branch - getting-started-example - instead.

https://devconnected.com/how-to-clone-a-git-repository/#Git_clone_a_specific_branch

remilamoureuxlevesque, 2022-05-27:
Thanks you, that work well.

remilamoureuxlevesque, 2022-05-27:
I got an other similar issue, after that I try to compile the capture exemple with the same method from the getting started example, but when I execute the `make -j4` command nothing append and the executable file is not genrated.

This is what I did. I know that I also run the example with the `rs-capture` command in my terminal, but I want to be able to add some modification to it.

```
remi@remi-desktop:~/Desktop/Realsense/librealsense/examples/capture$ ls
build  CMakeLists.txt  readme.md  rs-capture.cpp
remi@remi-desktop:~/Desktop/Realsense/librealsense/examples/capture$ cd build/
remi@remi-desktop:~/Desktop/Realsense/librealsense/examples/capture/build$ ls
remi@remi-desktop:~/Desktop/Realsense/librealsense/examples/capture/build$ cmake ../
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/remi/Desktop/Realsense/librealsense/examples/capture/build
remi@remi-desktop:~/Desktop/Realsense/librealsense/examples/capture/build$ make -j4
remi@remi-desktop:~/Desktop/Realsense/librealsense/examples/capture/build$ ls
CMakeCache.txt  CMakeFiles  cmake_install.cmake  Makefile
remi@remi-desktop:~/Desktop/Realsense/librealsense/examples/capture/build$
```

MartyG-RealSense, 2022-05-27:
rs-hello-realsense and the getting-started example that uses the same code are text-only programs.  **rs-capture** is a graphical program that requires graphics support to have been enabled when building the librealsense SDK.  When building the SDK from source code, support for graphics in examples and tools is included in the SDK build if the flag **-DBUILD_GRAPHICAL_EXAMPLES=true** is included in the CMake build instruction.  For example:

`cmake ../ -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

The need for this graphics support is referenced in the CMakeLists.txt file for the rs-capture program.

https://github.com/IntelRealSense/librealsense/blob/master/examples/capture/CMakeLists.txt#L8

remilamoureuxlevesque, 2022-05-27:
Oh ok I understand, but does the Debian package method already did that for me ? or I should rebuild from source with the needed flag ?

MartyG-RealSense, 2022-05-27:
I believe that when installing from packages, the graphical support is provided by the **librealsense2-utils** package of demos and tools that is one of the two required packages that is installed (the other being librealsense2-dkms)

MartyG-RealSense, 2022-06-02:
Hi @remilamoureuxlevesque  Do you require further assistance with this case, please?  Thanks!

remilamoureuxlevesque, 2022-06-02:
No everything is fine thanks for your help.

closed 2022-06-02 by remilamoureuxlevesque