realsenseai/librealsense #10552: L515 cannot be detected by the Laptop

sunwell1994, 2022-05-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Linux (Ubuntu 18/20) / MacOS  | 
| Kernel Version (Linux Only)    | 5.4.0-107-generic                                      | 
| Platform                                 | PC  |
| SDK Version                            |  {2.49 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

L515 performs differently on three different platforms.
1. It works well on my desktop. The system is ubuntu 20.04. It can work with USB 3.1 mode.
2. It cannot be detected on my Macbook now. It worked well 2 months ago. It suddenly couldn't work after one night before. 
3. It cannot be detected on my Linux laptop. The system is ubuntu 18.04. I tried different USBs to connect to the laptop. The original USB cannot be detected for most trials but it happened to be recognized once as the USB 2.0 mode.

MartyG-RealSense, 2022-05-29:
Hi @sunwell1994   If your L515 camera can function on an Ubuntu desktop and two Macbooks but not one other Macbook and an Ubuntu laptop then it sounds as though the camera hardware and its firmware driver are okay and it may be an issue with the USB ports or operating system on those two particular computers that it is not detected on.

As a starting point in investigating this case, please try unplugging the micro-sized end of the USB cable from the base of the L515, turning the connector around the other way and re-inserting it into the base of the camera (USB Type-C cables are two way insertion at the micro-sized end).  Does this resolve the problem, please?

In regard to the Macbook with the problem, which MacOS version is it using please?   If it has been upgraded to Monterey then there have been USB problems with RealSense camera detection that have only recently been solved by an installation guide at https://lightbuzz.com/realsense-macos/

sunwell1994, 2022-05-29:
> please try unplugging the micro-sized end of the USB cable from the base of the L515, turning the connector around the other way and re-inserting it into the base of the camera (USB Type-C cables are two way insertion at the micro-sized end). Does this resolve the problem, please?
It does not make L515 work.

> 
> In regard to the Macbook with the problem, which MacOS version is it using please? If it has been upgraded to Monterey then there have been USB problems with RealSense camera detection on that MacOS version that have only recently been solved by an installation guide at https://lightbuzz.com/realsense-macos/
All systems are Monterey. But 2 Macs can detect the Intel device by USB while the other one cannot find it at all.

MartyG-RealSense, 2022-05-29:
Has the camera ever worked on the 16.04 laptop or has it only just started being undetected on that PC overnight like with the one Macbook?

sunwell1994, 2022-05-29:
> Has the camera ever worked on the 16.04 laptop or has it only just started being undetected on that PC overnight like with the one Macbook?

Thanks for your reply. It has never worked on the Ubuntu 16.04 laptop.

MartyG-RealSense, 2022-05-29:
If lsusb cannot detect the camera on the 16.04 machine then it seems unlikely that it is a problem related to the librealsense SDK.  

You could test whether it is a conflict with the Linux kernel by building the SDK from source code with the **libuvc backend** method at the link below.  This build method is not dependent on Linux versions or kernel versions and does not require patching.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

sunwell1994, 2022-05-29:
> If lsusb cannot detect the camera on the 16.04 machine then it seems unlikely that it is a problem related to the librealsense SDK.
>
> You could test whether it is a conflict with the Linux kernel by [...]

I agree that the issue may not be related to the librealsense SDK. The laptop should detect the realsense device as Intel Corp at the USB port even without the SDK.  
I also tried the link but some downloads  cannot complete in the shell file.

sunwell1994, 2022-05-30:
> Does detection of the camera on the 16.04 machine change if you reload the **udev** device handling rules with the instruction below:
> 
> `sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules && sudo udevadm control --reload-rules && udevadm trigger`

It did not work as well.

MartyG-RealSense, 2022-05-30:
There are a small number of cases where a camera may work normally on one machine but never work on another and the exact reason why is never able to be found.  The Ubuntu 16.04 machine may fall into this category of unsolveable problems.  As the libuvc backend build failed despite its installation script having been originally designed for 16.04, this suggests that there may be a hidden issue with the computer (such as a glitch in the USB system) that is unrelated to librealsense.

The Macbook problem is unusual, since the camera worked normally until a particular day and it works with the other two Macbooks.  In 'it worked fine until today' cases, I would tend to suspect that an update may have been installed on the operating system of that particular computer that conflicted with the camera.

closed 2022-06-06 by sunwell1994