realsenseai/librealsense #10553: Extracting Depth matrix from raw in Python

singchee, 2022-05-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Win 11  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.50.0.3785                     | 
| Language                            |  Python                          | 
| Segment			|  Other                        | 

### Issue Description
Hi all

I would like to capture 2D single frames of faces, and extract a depth matrix into a CSV file for analysis offline. 

I have used the SDK 2D capture feature to get the RAW + PNG file. I would like to get the depth matrix into a CSV file from the RAW file. I have found sample code for this [here ](https://github.com/IntelRealSense/librealsense/issues/3632), but only for C++. 

Would anyone be able to point me to some resources to have this done in python (if possible?)? 

Thanks!

Regards
Sing

MartyG-RealSense, 2022-05-29:
Hi @singchee  There is little information about converting a .raw to .csv, and no librealsense references for doing so.  The few references that I did find though suggested doing so using a tool called **sed** to convert strings, like in the links below.

https://stackoverflow.com/questions/64947184/converting-very-large-raw-file-to-csv-file-tf-data-dataset

https://www.reddit.com/r/datasets/comments/27woaz/are_raw_files_convertible_to_csv_or_readable_in/

**************

Alternatively, the link below has some suggestions for code for a direct .raw to .csv conversion without using sed.

https://stackoverflow.com/questions/42469277/reading-and-splitting-a-raw-file-for-data-processing

MartyG-RealSense, 2022-06-05:
Hi @singchee  Do you require further assistance with this case, please?  Thanks!

singchee, 2022-06-06:
Thanks @MartyG-RealSense  

Haven't had a chance to play around with the python code yet, will update once i've had a go

singchee, 2022-06-06:
Thanks @MartyG-RealSense I've found it's a lot easier to just capture a .bag file at low frame rate, then convert to csv using the converter tool.

Could i check if the PNG file would contain the same data as a CSV file at the pixel level, using the underlying color gradients?

MartyG-RealSense, 2022-06-06:
When RealSense depth data is saved to a PNG image most of the depth information is lost, so reading depth values from a PNG is not accurate and it is better to use the **raw** image file format that preserves the depth information.  Details about accessing depth values from PNG or .raw can be found at https://github.com/IntelRealSense/librealsense/issues/8649#issuecomment-804792092

singchee, 2022-06-06:
Thanks , do you know if the CSV export via the converter loses detail as well?

MartyG-RealSense, 2022-06-06:
A csv file should not have the same loss of depth information during export as a PNG export as it is a **textual** format that can be loaded into databases and spreadsheets instead of an image format like PNG.

MartyG-RealSense, 2022-06-18:
Case closed due to no further comments received.

closed 2022-06-18 by MartyG-RealSense

JoeSmallman, 2022-12-13:
For anyone who wants some Python code:

```python
with open('depth.raw', 'rb') as f:
    img_data = f.read()
rows = 240
cols = 424
resolution = rows * cols
img = np.frombuffer(img_data, count=resolution, dtype=np.uint16)
img = img.reshape((rows, cols))
```

MartyG-RealSense, 2022-12-13:
Thanks so much @JoeSmallman for sharing your code with the RealSense community!