realsenseai/librealsense #10556: How to make a continous scan with the RealSense Viewer ?

Amayess, 2022-05-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D450 | 
| Firmware Version                   | 5.13.00.50 | 
| Operating System & Version |   Win 10  |                                     | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0 |
| Language                            |  C                           | 
| Segment			|  Robots    | 

### Issue Description
Hi @MartyG-RealSense,

I noticed while doing several scans with the RealSense viewer that it only takes into consideration the last frame, so the mesh I generate is not as good and clean even when applying the different post-processing filters.

My question: is there a way to make the RealSense viewer take into consideration a set of frames and do some kind of fusion between them in order to have a better rendering and mesh?

MartyG-RealSense, 2022-05-31:
Hi @Amayess  The RealSense SDK's **rs-kinfu (KinectFusion)** C++ example program performs frame fusion to assemble a point cloud scan in real-time as the camera is moved around and then can export a ply when you are satisifed with the level of detail on the image.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/kinfu

Amayess, 2022-05-31:
Okay, will do it ! 

Thank you :) !

Amayess, 2022-06-01:
Hi @MartyG-RealSense,

I have followed all the steps and compiled the required libraries with Cmake, the program compiles but when I do export to ply it generates a .ply file of 1KB instead of a larger file.

How can I fix this, please ? 

[image]

MartyG-RealSense, 2022-06-01:
If you open the **3D Viewer** tool of Windows and drag and drop the ply file into its window, is the model successfully loaded and displayed?

You can find and launch 3D Viewer by typing **3d viewer** into the text box at the bottom of your Windows screen, next to the button in the corner where the computer is shut down.

Here is a ply exported from the 3D mode of the RealSense Viewer that I dropped into 3D Viewer.

[image]

Amayess, 2022-06-01:
Any explication, please ? @MartyG-RealSense

MartyG-RealSense, 2022-06-01:
The image that you provided earlier shows that it is repeatedly printing **reset**.  According to the documentation for rs-kinfu, a "the algorithm attempts to fuse the current depth frame with the existing scene.  If it fails, the model obtained so far is reset and reconstruction of a new scene begins".

This suggests that the program might not be able to successfully export to ply a complete image because something in the scene is causing frame fusion to fail and reset the pointcloud.

RealSense users who try rs-kinfu typically experience an error with the line that attempts to load the **High Density** camera configuration preset file, and commenting out that line resolves the problem.  So it may be worth commenting out that line and rebuilding rs-kinfu as described at https://github.com/IntelRealSense/librealsense/issues/9463#issuecomment-885488381 to see whether this resolves the reset problem.

MartyG-RealSense, 2022-06-01:
The problem may possibly be if rs-kinfu is saving the ply in **binary** format.  By default the RealSense Viewer exports ply files as non-binary **textual** format.

[image]

In a non-RealSense guide about Kinfu at the link below, it also suggests that a ply should be saved without binary encoding in order for it to be used in MeshLab.

[image]

https://github.com/PointCloudLibrary/pcl/blob/master/doc/tutorials/content/using_kinfu_large_scale.rst#part-2-running-pcl_kinfu_largescale_mesh_output-to-convert-the-tsdf-cloud-into-a-mesh

MartyG-RealSense, 2022-06-01:
There is no documentation about the ply export feature of rs-kinfu, so there is no information about how the ply is formatted (other than apparently being binary).  

If you are missing normals, the guide for importing a RealSense ply into MeshLab explains in its **Creating Normals** section (located about a third of the way down the guide) how you can generate them in MeshLab after importing the ply by going to **Filters > Normals, Curvatures and Orientation > Compute normals for point sets** and then clicking the 'Apply' button and closing the window.

https://www.andreasjakl.com/capturing-3d-point-cloud-intel-realsense-converting-mesh-meshlab/

[image]

closed 2022-06-13 by MartyG-RealSense