realsenseai/librealsense #10558: use T265 in Raspberry pi 4

Ntstr, 2022-05-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   |  | 
| Operating System & Version |   Rasbian / linux 11  | 
| Kernel Version (Linux Only)    |       11 bullseye                                   | 
| Platform                                 | Raspberry Pi4  |
| SDK Version                            |                           | 
| Language                            |  python                           | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

I am trying to make a T265 work using python code on a raspberry pi 4 B running under rasbian
I had to do all the tutorials present on github before succeeding in downloading librerealsense and the pyrrealsense2 library
and now that I have succeeded realsense-viewer fails to detect my camera the same for the python program which shows me "No device connected"

the same for the rs-enumerate-devices command which does not return "No device connected"
however when I do lsusb the camera is well recognized but as "Intel Myriad VPU Movidius Neural Compute Stick"
I noticed while using the camera on windows that it was also recognized like this at the beginning but once I launched realsense viewer it became recognized as "intel realsense tracking camera t265" and my python programs work very well

**Here is my question how to make my t265 camera finally work on my raspberry pi4 and be able to use it in python programs?**

I admit that I am not an expert in raspberry and linux so I followed, start over, start over again the different tutorials trying to understand what I was doing until finally succeeding in installing librerealsense and pyrerealsense2 but it's still a hack between the different tutorials

if anyone can help me solve my problem I would be very grateful or lead me to another current tutorial for raspberry pi 4B

agennaro1, 2022-06-04:
I use Ubuntu Mate 20.10  was the only SO thats works for me without any problem

lycn2020, 2023-08-02:
I am using t265 on raspberry pi 4 4GB and ubuntu mate 20.10 arm64 and build librealsense 2.53.1 and when I run "realsense-viewer" not detect device . 
Help me please .

AlexSerbinov, 2023-12-17:
Have the same issue ..(

tak2hu, 2024-12-27:
I had the same "no device connected" every time I ran the binary I compiled even with different build options or checking out older versions.

The solution I encountered was copy pasting the binaries from an existing librealsense.so and it's executables, like from this [APSync image](https://firmware.ardupilot.org/Companion/apsync/apsync-rpi-ubuntu-t265-latest.img.xz) into the RPi 4 8 GB running Ubuntu 20.04, it works like a charm.

P.S: I tried using APSync directly, it wouldn't boot because it needed newer kernel versions, so I just copy pasted the boot folder from a working Ubuntu 20.04 into APSync and it boots and works with the t265, but when connecting to the Pixhawk by USB it errors because it can't find the usb ACM CDC kernel module, I tried fixing it and even compiling a kernel, nothing worked to connect the RPi 4 to the Pixhawk.

closed 2026-10-23 by MartyG-RealSense