realsenseai/librealsense #10562: Dependency Issues with ROS2 Wrapper

LiquidTurtle1, 2022-06-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i & L515  | 
| Operating System & Version |   Linux (Ubuntu 18.04LTS) Jetpack 4.5 | 
| ROS2 Distro | Foxy (Built from source) |
| Kernel Version                        |  4.9.140-tegra                                         | 
| Platform                                 | NVIDIA Jetson Xavier NX         |
| Architecture | Arm64 |
| SDK Version                            |  2.50.0                           | 

## Issue Description
Hello.
New to ROS2 and new to Linux, apologies for any basic questions or redundancies!

I am looking to implement SLAM using the D435i and L515 in a ROS2 environment. I have been able to run both the RealSense and ROS2 software independently, but I am encountering errors when I attempt to install the ROS2 Wrapper for RealSense. 

I installed the latest RealSense SDK with Debian Packages from the Jetson installation guide: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

### Installing Realsense ROS2 Wrapper:

```
jetson_dev@jetson-dev-desktop:~$ . ~/ros2_foxy/install/local_setup.bash
jetson_dev@jetson-dev-desktop:~$ rosdep install -i --from-path src --rosdistro $ROS_DISTRO --skip-keys=librealsense2 -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
realsense2_camera: No definition of [diagnostic_updater] for OS version [bionic]
realsense2_description: No definition of [xacro] for OS version [bionic]
```

if I choose to build anyway, regardless of the dependency issues

```
jetson_dev@jetson-dev-desktop:~$ . ~/ros2_foxy/install/local_setup.bash
jetson_dev@jetson-dev-desktop:~$ colcon build
Starting >>> realsense2_camera_msgs
[Processing: realsense2_camera_msgs]                             
Finished <<< realsense2_camera_msgs [41.0s]                       
Starting >>> realsense2_camera
Starting >>> realsense2_description              
Finished <<< realsense2_description [6.41s]                            
--- stderr: realsense2_camera                          
CMake Error at CMakeLists.txt:89 (find_package):
  By not providing "Findcv_bridge.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "cv_bridge", but CMake did not find one.

  Could not find a package configuration file provided by "cv_bridge" with
  any of the following names:

    cv_bridgeConfig.cmake
    cv_bridge-config.cmake

  Add the installation prefix of "cv_bridge" to CMAKE_PREFIX_PATH or set
  "cv_bridge_DIR" to a directory containing one of the above files.  If
  "cv_bridge" provides a separate development package or SDK, be sure it has
  been installed.

---
Failed   <<< realsense2_camera [7.49s, exited with code 1]

Summary: 2 packages finished [50.0s]
  1 package failed: realsense2_camera
  1 package had stderr output: realsense2_camera

```

I have previously tried installing on Jetpack 4.6.1, where I encountered the same issue. 

Do you have any suggestions?

MartyG-RealSense, 2022-06-04:
Hi @LiquidTurtle1  May I begin by asking the following questions, please:

1.  Did you build the librealsense SDK first and the RealSense ROS wrapper secondly after that?  This is the correct order that SDK and wrapper should be installed in.

2.  Did you follow the instruction steps for building the RealSense ROS2 wrapper at the link below in the order that the steps are listed in the instructions?

https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta#step-3-install-intel-realsense-ros2-wrapper-from-sources

LiquidTurtle1, 2022-06-04:
Hi Marty,
Thanks for the quick reply.

> 1. Did you build the librealsense SDK first and the RealSense ROS wrapper secondly after that?  This is the correct order that SDK and wrapper should be installed in.

Yes, I first installed the SDK with Debian Packages, and then the wrapper.

> 2. Did you follow the instruction steps for building the RealSense ROS2 wrapper at the link below in the order that the steps are listed in the instructions?

I did. When I ran `$ sudo rosdep init` I was given the following prompt:

```
ERROR: default sources list file already exists:
    /etc/ros/rosdep/sources.list.d/20-default.list
Please delete if you wish to re-initialize
```

and I did delete the source file, and ran the command again. `$ rosdep update` runs fine, 
but then `$ rosdep install -i --from-path src --rosdistro $ROS_DISTRO --skip-keys=librealsense2 -y` gave the error listed above.

I've noticed some other users had some success with their Xaver NX installing the ROS1 wrapper by using older versions of JetPack and older versions of the SDK. Might this be the same case for the ROS2 Wrapper? 

Thank you for any other suggestions you have. I will be unable to access my machine again until Monday, but I have a record of all of the steps that I took to get where I am.

MartyG-RealSense, 2022-06-04:
Thanks very much for the information.  If you are starting a fresh attempt at a source code build after a failed build then I would recommend completely deleting the **/ros2_ws/src** workspace folder instead of files within the folder and then create the /ros2_ws/src folder again on the next build attempt.  Did you delete the entire workspace folder when re-running the installation, please?

LiquidTurtle1, 2022-06-04:
Hi Marty,
Thanks for your questions and investing your time into helping me.

> Did you delete the entire workspace folder when re-running the installation, please?

I did. I actually started with a fresh JetPack image then reinstalled ROS2 and the SDK before trying again, but I ended up with the same error.

MartyG-RealSense, 2022-06-05:
My research of cases that experienced the same error with **sudo rosdep init**advised to remove rosdep with the command below and then re-enter the 'sudo rosdep init' command.

`sudo rm /etc/ros/rosdep/sources.list.d/20-default.list`

LiquidTurtle1, 2022-06-05:
Hi Marty,

> `sudo rm /etc/ros/rosdep/sources.list.d/20-default.list`

This is the approach I took, I'm glad to see this is the direction you would take as well.

My real issue is with  `rosdep install` . Do you have any suggestions for the output I received as described in my original post?

MartyG-RealSense, 2022-06-05:
Thanks very much for your patience.  I do not recall having come across the diagnostic_updater error before.  There was a recent case at https://github.com/IntelRealSense/realsense-ros/issues/2309 of the **No definition of [xacro] for OS version [bionic]** rosdep error though with ROS2 Foxy.  I suggested to try using **--ignore-src** in the rosdep instruction.

LiquidTurtle1, 2022-06-05:
Thanks for you feedback.

I believe I tried this on my previous installation, but I did not use the -r flag on my most recent install. I'll give that a try tomorrow when I have access to my Jetson again.

Thank you for all your help, I will report back tomorrow!

MartyG-RealSense, 2022-06-06:
If you prefer for the discussion to be on the RealSense ROS GitHub then I will close this case and we can continue at the RealSense ROS forum so that we are not discussing the same issue in two different locations.  Thanks!

closed 2022-06-06 by MartyG-RealSense