realsenseai/librealsense #10563: Help with deproject. Getting z coordinate same as distance

Arshad9544, 2022-06-04:
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.50.0 }                          | 
| Language                            |  {python}                          | 

### Issue Description
When I use :
distance = = depth_frame.get_distance(pixel_x, pixel_y)
point3D = rs.rs2_deproject_pixel_to_point(intrinsics,[pixel_x ,pixel_y],distance) why is the z coordinate same as the distance to object in all even if i move the object in different directions infront of the camera

MartyG-RealSense, 2022-06-04:
Hi @Arshad9544  It would be normal for Z to be approximately the same as the distance to camera if the camera was being moved up-down-left-right but not forward / backward, as the Z coordinate represents 3D depth (the distance of the camera from the observed object / surface).

Arshad9544, 2022-06-04:
Actually z coordinate should be representing the distance along the axis coming out of the camera and not the distance to the object. Am I wrong in this? I am working on a project which requires the x y z coordinates of the object. If the z axis always represents the depth or distance to object, is there anyway to get the x y z coordinates of the object in world coordinates with camera as origin. Thank you

MartyG-RealSense, 2022-06-04:
In RealSense the Z axis is the depth / distance from the front of the camera to an object.  

In ROS cases where a RealSense user requires distance to an object to be represented by a different axis such as Y instead of the default Z, it is possible to perform a **static transform** or to define a custom orientation in RViz (as described at the link below).

https://answers.ros.org/question/339109/how-to-change-the-cameras-view-in-rviz-in-the-begin/

In most applications, Z will be the distance from the camera to the object and I would recommend using Z if possible.

********

There a few different scripting approaches that could be used to obtain the XYZ world coordinates using Python without having to use ROS.  These include:

1.  Performing depth to color alignment and then obtaining the XYZ coordinates by 'deprojecting' from 2D to 3D with the SDK instruction **deproject_pixel_to_point**.  An example of this approach with Python is at https://github.com/IntelRealSense/librealsense/issues/6749#issuecomment-654185205

2.  Generate a 3D point cloud and then obtain its vertices with the **points.get_vertices** instruction and store them in a numpy array so the XYZ coordinates can be retrieved from the array.

Differences between approaches 1 and 2 are discussed with RealSense team members at https://github.com/IntelRealSense/librealsense/issues/4315

3.  Use the **project_color_pixel_to_depth_pixel** to convert a single specific color pixel to a depth pixel to obtain its XYZ without having to align the entire image.  https://github.com/IntelRealSense/librealsense/issues/9945#issuecomment-972539640 may be a useful Python reference for that approach.

closed 2022-06-18 by MartyG-RealSense