realsenseai/librealsense #10564: D455

sosoyeon1125, 2022-06-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400-D455 } | 
| Firmware Version                   | (05.13.00.50) | 
| Operating System & Version |   { Linux (Ubuntu 18.04.6) } | 
| Kernel Version (Linux Only)    |  5.4.0-113-generic                                       | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.50.0 }                          | 
| Language                            |  {python}                          | 
| Segment			|  {others }                          | 

### Issue Description
< question >

Hello, I am a student using D455. I'm asking you a question because there are many difficulties because it's my first time using Ubuntu& realsense. 

What I want to create is an algorithm that measures a person's height. 

I want to take the value of a specific height and determine who matches the value of that height. 

May I know if there is an example that already exists? 

Is there an example of measuring the size of an object(person)?

Is there an example  found of depth value to find a match to that  depth value of external input?

sosoyeon1125, 2022-06-04:
Platform -> pc Ubuntu

MartyG-RealSense, 2022-06-04:
Hi @sosoyeon1125  As you are using Python, the RealSense SDK's **TensorFlow** wrapper may suit your needs as it has an example program for measuring the height of a person displayed on an RGB image.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/tensorflow#part-2---augmenting-output-using-depth-data

As the example deals with minimum and maximum height, I would speculate that you could adapt the example to not display height information for someone whose height did not fit within a defined min-max range.

sosoyeon1125, 2022-06-04:
oh. TensorFlow !!!

sosoyeon1125, 2022-06-04:
and "Is there an example found of depth value to find a match to that depth value of external input?" <- this question, can i konw the answer?

sosoyeon1125, 2022-06-04:
[image]

[INFO] start streaming...
Traceback (most recent call last):
  File "person_height.py", line 17, in <module>
    pipeline.start(config)
RuntimeError: Couldn't resolve requests
-> Is there any workaround for this problem?

MartyG-RealSense, 2022-06-04:
> As the example deals with minimum and maximum height, I would speculate that you could adapt the example to not display height information for someone whose height did not fit within a defined min-max range.

I addressed this question in the above quote.  You could likely set the program so that it only provides height information for people of a specific height.  Matching height to a specific person would be more complicated though and probably involve some form of database to store the heights of each scanned individual.  There is not an existing RealSense example program for this.

You could conceivably use TensorFlow to compare distances between body 'pose' joints to tell the difference in height between peoples' head and feet, but again this does not provide the identities of specific people.

https://blog.tensorflow.org/2018/05/real-time-human-pose-estimation-in.html

MartyG-RealSense, 2022-06-04:
In regard to the **Couldn't resolve requests** error, the 'person_height.py' script tries to start depth and color streams with resolution 848x480, as defined on lines 6 and 7. 

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/tensorflow/example2%20-%20person%20height.py#L6-L7

If the D455 is detected as being on a USB 2 connection instead of USB 3 then only 5 or 10 FPS would be supported for the depth stream instead of the requested 30.  The 848x480 resolution would also be completely unsupported for the color stream.

If you change W to 640 and H to 480 to set a 640x480 resolution for both streams then this should enable the streams to be enabled successfully if the problem is indeed related to the camera being detected as being on USB 2.

```
W = 640
H = 480
```

closed 2022-07-28 by sosoyeon1125