realsenseai/librealsense #10567: Trying to build custom code to control GPIO pins

JoshuaSummers, 2022-06-06:
Hey! I was only able to get the librealsense sdk installed after I flashed my SD card with a pre-built image using the Open-Source Ethernet tutorial. I'm using a raspberry pi 3b. 

I want to use the hello-realsense code to turn on an LED if the camera is facing an object less than 1 meter away. I need to use wiringPi to control the gpio pins, so I feel like this complicates thing. 

Here's my code:
#include <wiringPi.h>
#include <string>
#include <librealsense2/rs.hpp>
rs2::pipeline p;
p.start();
... (it's just the rest of the rs-hello-realsense.cpp code, then it's:)
int main(void)
wiringPiSetup();
pinMode(0, OUTPUT);
digitalWrite(0, LOW);
while(dist_to_center<1)
{
digitalWrite(0, HIGH);
} return 0; }

However, when I type "gcc -Wall -o Blink Blink.cpp -lwiringPi", I get the error "Blink.cpp:11:1: error: 'p' does not name a type" for p.start();

I know this is because I made my own folder not in librealsense, and I am supposed to use something like the getting-started template (#2946). However, I couldn't find the getting-started folder, and I think I need to execute the command with "gcc -lwiringPi", or else I won't be able to control the gpio pins. I could be wrong about that though, but when I don't include -lwiringPi I can't use wiringPi.

Seriously appreciate the help.

MartyG-RealSense, 2022-06-07:
Hi @JoshuaSummers  If you are using Ubuntu on your Pi 3b and you do not require the ethernet networking support included in the pre-built SD card image then you could try the simple **libuvc backend** installation method in the link below that has been recommended by Intel for Pi 3 in the past.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

Intel also provide a Pi 3b guide for installing librealsense on the **Ubuntu MATE** OS.

https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md

In regard to your question, are you referring to the **rs-getting-started.cpp** project referenced at https://github.com/IntelRealSense/librealsense/issues/10550#issuecomment-1139347185 that is essentially the rs-hello-realsense script but with a CMakeLists.txt build file designed for finding the librealsense library wherever it is located?

Also, can you confirm whether the program is able to run if **-lwiringPi** is left off the launch command or whether the error still occurs?

It sounds as though you are attempting to control an LED attached to the GPIO pins of the Raspberry Pi board.  Is that correct, please?

JoshuaSummers, 2022-06-07:
> Hi @JoshuaSummers If you are using Ubuntu on your Pi 3b and you do not require the ethernet networking support included in the pre-built SD card image then you could try the simple **libuvc backend** installation method in the link below that has been recommended by Intel for Pi 3 in the past.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md
> 
> Intel also provide a Pi 3b guide for installing librealsense on the **Ubuntu MATE** OS.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md
> 
> In regard to your question, are you referring to the **rs-getting-started.cpp** project referenced at [#10550 (comment)](https://github.com/IntelRealSense/librealsense/issues/10550#issuecomment-1139347185) that is essentially the rs-hello-realsense script but with a CMakeLists.txt build file designed for finding the librealsense library wherever it is located?
> 
> Also, can you confirm whether the program is able to run if **-lwiringPi** is left off the launch command or whether the error still occurs?
> 
> It sounds as though you are attempting to control an LED attached to the GPIO pins of the Raspberry Pi board. Is that correct, please?

Hey Marty!! Thanks for responding. I tried the Ubuntu MATE guide, but Ubuntu MATE was too much for my rpi to handle. The libuvc backend was taking too long, though if for some reason I need to reset my progress I'd probably use the libuvc method. Right now, I think I'm fine and I don't think I have to do anything regarding this as of now.

Yes, I am attempting to control an LED attached to the GPIO pins of the Raspberry Pi board. I'm only using C++ now because most of the examples were in C++, and the C++ code seemed easy to work with. I could use python, as controlling the gpio pins is easier that way on the raspberry pi. I didn't get any errors when I left -lwiringPi off the launch command, sorry for not checking that in advance. And yes, I am referring to the rs-getting-started.cpp project with the special CMakeLists.txt build file. Appreciate the help!

MartyG-RealSense, 2022-06-07:
If Python is your preference then a pyrealsense2 script at https://github.com/IntelRealSense/librealsense/issues/6749#issuecomment-654185205 for printing the distance from the camera to a given pixel may provide similar results to the C++ rs-hello-realsense.

JoshuaSummers, 2022-06-07:
> 

Yeah, python seems to be the answer! Now I just need to debug code and get things working. I have one more question though: I copied the code from that thread, and when I have pipeline.start(config) in my code, I get RuntimeError: Couldn't resolve requests. Do you know how the fix this? Thanks for your quick responses.

MartyG-RealSense, 2022-06-07:
**Couldn't resolve requests** typically means that a stream configuration was requested by the script that was not supported by your particular camera model at the time that the script is run.   Which RealSense camera model are you using, please?

JoshuaSummers, 2022-06-07:
I'm using the D435!

MartyG-RealSense, 2022-06-07:
As you are using Pi 3B, it will have a USB 2 port as USB 3 was not introduced until Pi 4.  So the range of supported resolution / FPS modes on USB 2 will be more limited than a USB 3 connection.

It should work on your Pi 3B and D435 if you reduce the FPS in the script from 30 to 6 for the 1280x720 resolution.

```
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 6)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 6)
```

If you require 30 FPS, try this:

```
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
```

JoshuaSummers, 2022-06-07:
> As you are using Pi 3B, it will have a USB 2 port as USB 3 was not introduced until Pi 4. So the range of supported resolution / FPS modes on USB 2 will be more limited than a USB 3 connection.
>
> It [...]

My code works and I'm able to turn on the LED!! I was super super happy, thanks a bunch <3. You're amazing.

closed 2022-06-07 by JoshuaSummers

MartyG-RealSense, 2022-06-07:
That's great news - thanks very much for the update and your kind words.  :)