realsenseai/librealsense #10568: How to query the external parameters from camera to imu

canglangzhige, 2022-06-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)
* 
Hello, I run ./realsense-viewer, and I can find stereo baseline/ right intrinsics/left intrinsics in Calibration data  and so on. As shown in the following figure:
[image]
But there is no coordinate system transformation of the camera relative to the imu. How can I query the factory parameters of the camera's coordinate system transformation relative to the imu?
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   |05.13.00.50| 
| Operating System & Version |   Linux (Ubuntu 20)  | 
| Kernel Version (Linux Only)    |  (e.g. 5.13.0-41)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.0 }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2022-06-07:
Hi @canglangzhige  You can launch the RealSense SDK's 'rs-enumerate-devices' tool in calibration information mode with the command **rs-enumerate-devices -c** to list the extrinsics of the camera that include the translation between different sensors (including gyro and depth, accel and depth, etc).

[image]

[image]

canglangzhige, 2022-06-07:
> Hi @canglangzhige You can launch the RealSense SDK's 'rs-enumerate-devices' tool in calibration information mode with the command **rs-enumerate-devices -c** to list the extrinsics of the camera that include the translation between different sensors (including gyro and depth, accel and depth, etc).
> 
> The information shown below was retrieved from my D455 camera model. You should query your own camera to find its unique extrinsic values.
> 
> [image]
> 
> [image]

Thanks for your answer！
I combined gyro and accel with linear_interpolation in the file rs_camera.launch:
[image]
I have launched the calibration information with the command rs-enumerate-devices -c. 
Extrinsic from "Infrared 2" To	"Accel"  is the same with Extrinsic from "Infrared 2" To	"Gyro" :
[image]
Can I assume that both parameters are Extrinsic from "Infrared 2" To "IMU" ?
In addition, what are the steps for automatic calibration of realsense d455? 
The parameters I calibrated by the Kalibr method are very different from the factory parameters. How much will the automatic calibration differ from the factory parameters? 
If I don't calibrate and use the factory parameters directly, will it have a big impact on my results?

MartyG-RealSense, 2022-06-07:
Accel and Gyro both come from a single IMU component inside the camera, so it makes sense that they would both have the same physical position in relation to other sensors.

Any time that the transform between two sensors is being described, it is an Extrinsic instead of an Intrinsic.

The D455 camera model has a compensation mechanism intended to mitigate the effect of thermal propagation in optics.  When active (default = On) it will track and adjust Depth and RGB calibration parameters automatically. 

There is no way to tell in advance how automatic calibration would change the values in comparison to the original factory-default calibration values.

As you are already familiar with using the camera calibration data interface in the RealSense Viewer, it may be worth using the guide at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 to reset the camera to its factory defaults using this interface and write it to the camera hardware to see if it works as well for you as performing a re-calibration.

canglangzhige, 2022-06-07:
> Accel and Gyro both come from a single IMU component inside the camera, so it makes sense that they would both have the same physical position in relation to other sensors.
> 
> Any time that the transform between two sensors is being described, it is an Extrinsic instead of an Intrinsic.
> 
> The D455 camera model has a compensation mechanism intended to mitigate the effect of thermal propagation in optics. When active (default = On) it will track and adjust Depth and RGB calibration parameters automatically.
> 
> There is no way to tell in advance how automatic calibration would change the values in comparison to the original factory-default calibration values.
> 
> As you are already familiar with using the camera calibration data interface in the RealSense Viewer, it may be worth using the guide at [#10182 (comment)](https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487) to reset the camera to its factory defaults using this interface and write it to the camera hardware to see if it works as well for you as performing a re-calibration.

As you said " When active (default = On) it will track and adjust Depth and RGB calibration parameters automatically." Can Infrared 1 and Infrared 2 be automatically calibrated?
Four other questions:
1. How do I turn off infrared when using ROS？
2. Why is the parameter Coeffs of Infrared 1/2 all 0, is the Infrared 1 or Infrared 2 I get undistorted? :
[image]
Why is this parameter not all 0 for color images?
[image]
3. Topic "Infrared" refers to what video stream？
4. Is there any documentation detailing these issues?

canglangzhige, 2022-06-07:
> 
1. But the documentation says that the depth is generated by the left and right cameras, and the IR frames just increase its accuracy. 
[image]
3. If I only use the left and right cameras (using the d455 as a stereo camera), will the IR frames affect the accuracy of the left and right cameras because I see a lot of laser dots on the left and right images?
4. How to turn off IR frames when using ROS？
5. Are you saying that "enable_infra" and "enable_infra 1" in the parameter are the same?
6. How to get color left and right images? I changed the parameter below to "true" but it didn't work. 
[image]
   If I use the following set of parameters, the topics only have "infra1" and "infra2", and no "infra":
[image]
  If I set "infra_rgb" to "true", then only colored "infra":
[image]
  If I set "enable_infra" to "false" and "infra_rgb" to "true", then there is no talk about "infra":
[image]
 What is the relationship between these parameters, and how can I get a color image of the left and right cameras?

canglangzhige, 2022-06-07:
OK. Thank you for your patience.

closed 2022-06-20 by MartyG-RealSense