realsenseai/librealsense #10570: L515 map_device_descriptor Cannot open /dev/video Last Error No such device

rancheng, 2022-06-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |  Ubuntu18.04 | 
| Kernel Version (Linux Only)    | 5.4.0-113-generic                                      | 
| Platform                                 | PC  |
| SDK Version                            |  2.50.0-0~realsense0.6127                        | 
| Language                            | C++                    | 
| Segment			|  Robot                       | 

### Issue Description

I was trying to retrieve `color image` and `imu` data to do camera-imu calibration. I wrote my code based on `TUM-basalt`, you can find the code from here:

[https://github.com/rancheng/basalt-mirror/blob/master/src/rs_l515_record.cpp](https://github.com/rancheng/basalt-mirror/blob/master/src/rs_l515_record.cpp)

When run the program, it always stuck at first frame and prompt the following error:

```shell
 08/06 11:43:00,485 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,487 ERROR [140013064918784] (sensor.cpp:572) acquire_power failed: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,488 ERROR [140013064918784] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,588 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,588 ERROR [140013064918784] (sensor.cpp:572) acquire_power failed: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,588 ERROR [140013064918784] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,688 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,688 ERROR [140013064918784] (sensor.cpp:572) acquire_power failed: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,688 ERROR [140013064918784] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,789 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,789 ERROR [140013064918784] (sensor.cpp:572) acquire_power failed: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,789 ERROR [140013064918784] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,889 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,889 ERROR [140013064918784] (sensor.cpp:572) acquire_power failed: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,889 ERROR [140013064918784] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,989 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,989 ERROR [140013064918784] (sensor.cpp:572) acquire_power failed: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,989 ERROR [140013064918784] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:01,090 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:01,090 ERROR [140013064918784] (sensor.cpp:572) acquire_power failed: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:01,090 ERROR [140013064918784] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:01,190 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
```

I tried to use `INFRARED` only, and made sure the `config` for streams are all valid.

you can follow the `basalt` official build instruction and install `gcc9` to build the project:

```shell
sudo apt-get update && apt-get install -y wget gnupg lsb-release software-properties-common && apt-get clean all
sudo apt-get update && sudo apt-get install -y wget gnupg lsb-release software-properties-common && apt-get clean all
sudo apt-get update && sudo apt-get install -y wget gnupg lsb-release software-properties-common
wget -O - https://apt.llvm.org/llvm-snapshot.gpg.key| apt-key add -
sudo wget -O - https://apt.llvm.org/llvm-snapshot.gpg.key| apt-key add -
sudo wget -O - https://apt.llvm.org/llvm-snapshot.gpg.key| sudo apt-key add -
echo "deb http://apt.llvm.org/bionic/ llvm-toolchain-bionic-11 main" > /etc/apt/sources.list.d/llvm11.list
sudo echo "deb http://apt.llvm.org/bionic/ llvm-toolchain-bionic-11 main" > /etc/apt/sources.list.d/llvm11.list
sudo echo "deb http://apt.llvm.org/bionic/ llvm-toolchain-bionic-12 main" > /etc/apt/sources.list.d/llvm12.list
sudo echo "deb http://apt.llvm.org/bionic/llvm-toolchain-bionic-11 main" > /etc/apt/sources.list.d/llvm11.list
sudo echo "deb http://apt.llvm.org/bionic/ llvm-toolchain-bionic-11 main" >> /etc/apt/sources.list.d/llvm11.list
sudo gedit /etc/apt/sources.list.d/llvm11.list
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u
sudo apt-get install -y cmake git libtbb-dev libeigen3-dev libglew-dev ccache libjpeg-dev libpng-dev openssh-client liblz4-dev libbz2-dev libboost-regex-dev libboost-filesystem-dev libboost-date-time-dev libboost-program-options-dev libopencv-dev libpython2.7-dev libgtest-dev lsb-core gcovr ggcov lcov librealsense2-dev librealsense2-gl-dev librealsense2-dkms librealsense2-utils doxygen graphviz libsuitesparse-dev clang-11 clang-format-11 clang-tidy-11 clang-12 clang-format-12 clang-tidy-12 g++-9 libfmt-dev && apt-get clean all
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-9 60 --slave /usr/bin/g++ g++ /usr/bin/g++-9
gcc --version
g++ --version
```

Then build the project

```shell
git clone --recursive https://github.com/rancheng/basalt-mirror.git
cd basalt-mirror
./scripts/install_deps.sh
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo
make -j8
```

To run the program please do the following:

```shell
basalt_rs_l515_record --dataset-path ~/l515_calib_data/ --manual-exposure 
```

rancheng, 2022-06-08:
This is the full running log of the program and a gui attached:

[image]

```shell
/home/ran/CLionProjects/basalt-mirror/cmake-build-debug/basalt_rs_l515_record --dataset-path /home/ran/basalt_ds/ --manual-exposure
Sensor L500 Depth Sensor. Supported options:
	Visual Preset
	Frames Queue Size
	Error Polling Enabled
	Depth Units
	Inter Cam Sync Mode
	LDD temperature
	Mc Temperature
	Ma Temperature
	Global Time Enabled
	Apd Temperature
	Depth Offset
	Freefall Detection Enabled
	Sensor Mode
	Host Performance
	Humidity Temperature
	Enable Max Usable Range
	Alternate IR
	Noise Estimation
	Enable IR Reflectivity
	Digital Gain
	Laser Power
	Confidence Threshold
	Min Distance
	Receiver Gain
	Post Processing Sharpening
	Pre Processing Sharpening
	Noise Filtering
	Invalidation Bypass
Sensor RGB Camera. Supported options:
	Backlight Compensation
	Brightness
	Contrast
	Exposure
	Gain
	Hue
	Saturation
	Sharpness
	White Balance
	Enable Auto Exposure
	Enable Auto White Balance
	Frames Queue Size
	Power Line Frequency
	Auto Exposure Priority
	Global Time Enabled
	Host Performance
Sensor Motion Module. Supported options:
	Frames Queue Size
	Enable Motion Correction
	Global Time Enabled
Device Intel RealSense L515 connected
Enabling manual exposure control
got calib
Skipping gyro data. Timestamp before the first accel measurement.Skipping gyro data. Timestamp before the first accel measurement.Skipping gyro data. Timestamp before the first accel measurement.frame set size: 1
frame count: 1
Timestamp / exposure 0: 152123618000 / 499ms
done img data reading ...
 08/06 11:43:00,485 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,487 ERROR [140013064918784] (sensor.cpp:572) acquire_power failed: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,488 ERROR [140013064918784] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,588 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,588 ERROR [140013064918784] (sensor.cpp:572) acquire_power failed: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,588 ERROR [140013064918784] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
 08/06 11:43:00,688 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such device
[... 38 lines cut ...]
 08/06 11:43:01,991 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 08/06 11:43:01,991 ERROR [140013064918784] (sensor.cpp:572) acquire_power failed: map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 08/06 11:43:01,991 ERROR [140013064918784] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 08/06 11:43:02,091 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 08/06 11:43:02,091 ERROR [140013064918784] (sensor.cpp:572) acquire_power failed: map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 08/06 11:43:02,091 ERROR [140013064918784] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 08/06 11:43:02,191 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 08/06 11:43:02,191 ERROR [140013064918784] (sensor.cpp:572) acquire_power failed: map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 08/06 11:43:02,191 ERROR [140013064918784] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 08/06 11:43:02,292 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 08/06 11:43:02,292 ERROR [140013064918784] (sensor.cpp:572) acquire_power failed: map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 08/06 11:43:02,292 ERROR [140013064918784] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 08/06 11:43:02,392 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 08/06 11:43:02,392 ERROR [140013064918784] (sensor.cpp:572) acquire_power failed: map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 08/06 11:43:02,392 ERROR [140013064918784] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
[... cut ...]
 08/06 11:43:02,692 ERROR [140013064918784] (librealsense-exception.h:52) map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 08/06 11:43:02,692 ERROR [140013064918784] (sensor.cpp:572) acquire_power failed: map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
 08/06 11:43:02,692 ERROR [140013064918784] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: map_device_descriptor Cannot open '/dev/video0 Last Error: No such file or directory
```

MartyG-RealSense, 2022-06-08:
Hi @rancheng  Could you please first try unplugging the micro-sized end of the USB cable from the base of your L515 camera, turning the connector around the opposite way and re-inserting it into the base of the camera (USB-C cables are two-way insertion at the micro-sized end).  Do you still experience the errors with the connector reversed, please?

rancheng, 2022-06-08:
I have also commented out the `Pangolin` side to diagnose the issue part.

The problem happens immediately after first color frame was read.

rancheng, 2022-06-08:
I also tried to do nothing in callback just print the strings it also prompt this error:

```cpp
else if (frame.get_profile().stream_type() == RS2_STREAM_COLOR) {
      std::cout << "color frame ..." << std::endl;
      /*
      ...
      */
}
```

rancheng, 2022-06-08:
The strange thing is that I can access the exposure and sensor data from python script with `pyrealsense2` library without any problem.

```python
def callback(frame):
    global color_frame_data, depth_frame_data, gyro_frame_data, accel_frame_data
    pr = frame.get_profile()
    if pr.stream_type() == rs.stream.gyro:
        # print("got gyro frame ...")
        pass
    if pr.stream_type() == rs.stream.accel:
        # print("got accel frame ...")
        pass
    if pr.stream_type() == rs.stream.color:
        print("got color frame ...")
    if frame.is_frameset():
        frameset = frame.as_frameset()
        f1 = frameset.get_color_frame().as_video_frame()
        f2 = frameset.get_depth_frame().as_video_frame()
        color_data = np.asanyarray(f1.get_data())
        print("color_data.shape: {}".format(color_data.shape))
        depth_data = np.asanyarray(f2.get_data())
        print("depth_data.shape: {}".format(depth_data.shape))
        actual_exposure = f1.get_frame_metadata(rs.frame_metadata_value.actual_exposure) * 1e-6
        print("actual_exposure: {}".format(actual_exposure))
```

The output is expected:

```shell
color_data.shape: (720, 1280)
depth_data.shape: (480, 640)
actual_exposure: 0.149205
color_data.shape: (720, 1280)
depth_data.shape: (480, 640)
actual_exposure: 0.149205
color_data.shape: (720, 1280)
depth_data.shape: (480, 640)
actual_exposure: 0.149205
color_data.shape: (720, 1280)
depth_data.shape: (480, 640)
actual_exposure: 0.149205
color_data.shape: (720, 1280)
depth_data.shape: (480, 640)
actual_exposure: 0.149205
```

rancheng, 2022-06-08:
Hi @MartyG-RealSense , Thanks for your advice, I closed the depth and only enabled IMU and color sensors streams, it works!

Also I followed the advice from [#issue50](https://gitlab.com/VladyslavUsenko/basalt/-/issues/50#note_704789113).

Removing the `device.hardware_reset()` line in my code seems to fix this bug.

I am going to close this issue as it is solved.

Thanks again @MartyG-RealSense !

closed 2022-06-08 by rancheng