realsenseai/librealsense #10571: How to align the depth and RGB frame from time dimension

seliber, 2022-06-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 14/16/17) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | imx8  |
| SDK Version                            |  2.50                          | 
| Language                            |  C++                         | 
| Segment			| Others                          | 

### Issue Description

Recently, we found an issue about the alignment for depth and RGB image. the RGB frame was about 1 second later than the depth frame. 

Do we need to set any configuration to align the depth and RGB image?

the API used:

	_config.enable_stream(RS2_STREAM_COLOR, _colorWidth, _colorHeight, RS2_FORMAT_BGR8, 15);
	_config.enable_stream(RS2_STREAM_DEPTH, _depthWidth, _depthHeight, RS2_FORMAT_Z16, 15);
	_config.can_resolve(_pipeline);
	_pipeline.start();
	//use try_wait_for_frames() to get frame
	auto frameSet;
	_pipeline.try_wait_for_frames(&frameSet, 200);

MartyG-RealSense, 2022-06-08:
Hi @seliber  When using a cfg / config definition, it should be placed within the brackets of the pipeline start instruction, otherwise the program will not know it should be using your custom configuration and will ignore it and use the camera's default configuration instead.

**_pipeline.start(_config);**

seliber, 2022-06-08:
Thanks @MartyG-RealSense , sorry, that's a typo. 
we use _pipeline.start(_config);

MartyG-RealSense, 2022-06-08:
Thanks for the clarification.  How does it behave if you do not put any instructions in the try_wait_for_frames bracket?

**_pipeline.try_wait_for_frames();**

seliber, 2022-06-08:
then how can I get the frame result?

MartyG-RealSense, 2022-06-08:
try_wait_for_frames is used to return a 'false' status if there is a timeout instead of allowing the program to throw an error and exit, as described at https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709

seliber, 2022-06-08:
actually , we monitor the result, it is always true. 
btw, do you mean to use **_pipeline.wait_for_frames()**?

MartyG-RealSense, 2022-06-08:
Most people will use wait_for_frames() instead of try_wait_for_frames(), yes

seliber, 2022-06-08:
Thanks, I'll try then.

seliber, 2022-06-08:
@MartyG-RealSense Thanks for the advice, I have tried to change the API, but no lucky to get the correct frame. The RGB frame was still later than the depth frame.

MartyG-RealSense, 2022-06-08:
https://github.com/IntelRealSense/librealsense/issues/1548#issuecomment-389070550 advises that there will be a temporal offset between depth and RGB frames but the wait_for_frames() instruction should be minimizing this by finding the best match between frame timestamps.  This is assuming that **frame drops** are not occurring though.

You could try enforcing a constant FPS rate for both streams instead of permitting FPS to vary in order to test whether there is an FPS lag on the RGB stream.  This can be done by having auto-exposure enabled and an RGB option called **Auto-Exposure Priority** disabled.  C++ code for disabling auto-exposure priority on the RGB sensor is provided below.

```
auto sensor = pipe_profile.get_device().query_sensors()[1];
sensor.set_option(RS2_OPTION_AUTO_EXPOSURE_PRIORITY, 0);
```

MartyG-RealSense, 2022-06-08:
Setting the stream to a certain value in the config instructions does not guarantee that it will actually remain constantly at that value unless the FPS is forced deliberately to remain constant, otherwise FPS can drift to a slower value.

seliber, 2022-06-08:
I tried to disabled auto-exposure, but still the two frames are not matched.

seliber, 2022-06-08:
I mean I use the way you told as below:
auto sensor = pipe_profile.get_device().query_sensors()[1];
sensor.set_option(RS2_OPTION_AUTO_EXPOSURE_PRIORITY, 0);

And I also get the original timestamp, the diff from RGB to Depth is large than 1000. 

I'll enable BUILD_EXAMPLE and try again. Thanks for your help.

seliber, 2022-06-09:
Sorry @MartyG-RealSense , we don't have UI output on the imx8 board, so that I cannot build or run "rs-align" example.

seliber, 2022-06-09:
@MartyG-RealSense , just FYI, I re-compile the camera SDK with the option -DBUILD_NETWORK_DEVICE=true, then the RGB delay is back to normal. 

but I didn't use any network features, just fetch the frame, why do I need to build with this option, it made the size of the libary increased

MartyG-RealSense, 2022-06-09:
There is not an obvious reason why using the ** -DBUILD_NETWORK_DEVICE=true** flag would have an effect on RGB on a system that is not using networking.

If you do not have UI output, does that mean that you are using a 'headless' setup without a display?  If you are then I would recommend specifying in your librealsense build that it is a headless'system that does not require graphics support to be included in the build by using the CMake build terms ** -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=false**

By setting DBUILD_GRAPHICAL_EXAMPLES=false then only text-based example programs and tools will be built and OpenGL-based graphical tools and examples such as realsense-viewer will be excluded.

seliber, 2022-06-10:
`cmake ../. -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=ON **-DBUILD_NETWORK_DEVICE=true**  -DBUILD_GRAPHICAL_EXAMPLES=false -DBUILD_CV_EXAMPLES=false -DBUILD_CV_KINFU_EXAMPLE=false -DBUILD_GLSL_EXTENSIONS=false  -DCMAKE_TOOLCHAIN_FILE=~/mycompiler.cmake  `

This is my compile option.

MartyG-RealSense, 2022-06-10:
I cannot see anything in the above CMake build instruction that would clearly suggest why you are having problems with alignment.  You could try a simpler build instruction though to eliminate the possibility that one of the flags is causing the problem, as some of the settings that were being set to false are already false by default anyway and I am not aware of a need to use DCMAKE_TOOLCHAIN_FILE unless building headless for the RealSense Android wrapper.

DBUILD_EXAMPLES is True by default, so if you want to exclude the headless textual tools then set it to False.

**cmake ../. -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=ON -DBUILD_EXAMPLES=false**

closed 2022-06-22 by MartyG-RealSense