realsenseai/librealsense #10575: Focal length of the realsense D435

monsterITC, 2022-06-08:
| Camera Model                       | {D435} | 
| Firmware Version                   | V2.50.0 | 
| Operating System & Version |   Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  4.9.253                                       | 
| Platform                                 | Nvidia Jetson xavier  |
| Language                            |  {Python}                          | 
| Segment			|  {Robot }                          | 

With the two IR cameras from the real sense, I can get the disparity. But when I want to convert it the depth image, It seems like it doesn't give me the correct depth. 

I use depth = (baseline*focal_length)/disparity

where baseline = 50mm = 0.05m
          focal_length= 942.8px, this one I am not, but I saw a guy comment it is this value. 

when I use this value of the focal_length, the depth seem too far, when I convert it to the point cloud, 
but when I random choose the value of 300, It seems gives me a better result. 

So I'm not sure if the problem come from this one or another thing that I miss about?

monsterITC, 2022-06-09:
Thank you very much for the information @MartyG-RealSense. I founded it.

closed 2022-06-09 by monsterITC

MartyG-RealSense, 2022-06-09:
You are very welcome, @monsterITC - thanks very much for the update!

by011, 2022-11-10:
> Hi @monsterITC You can find the focal length for your particular camera at a particular resolution using the calibration information mode of the RealSense SDK's 'rs-enumerate-devices' tool. The tool can be launched in calibration mode with the command **rs-enumerate-devices -c**
> 
> If you look at the **intrinsics** for a particular resolution and stream type then it will list 'fx' and 'fy'. This value is the focal length. Using either fx or fy will be fine as they are the same value.

Sorry for disturbing, I came across the similar prob, I searched for "Intrinsic of 'Color' / 640x480 ....", the Fx value is 608.6419, while Fy is 607.1061, which should i consider?

MartyG-RealSense, 2022-11-10:
Hi @by011  It's no trouble at all  )

On my own D435, for the 640x480 color intrinsics profile Fx is 607.5303 and Fy is 607.0724 (each camera unit has its own unique intrinsics determined during manufacturing at the factory).  Whilst Fx and Fy are commonly the same value, they are allowed to be different.  So both values are valid.  The focal length of the image will be calculated from a multiplication of pixel width by pixel height: Fx x Fy.

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#intrinsic-camera-parameters

MartyG-RealSense, 2025-06-19:
Hi @Myat2005  An Intel guide to stereo depth at the link below has a Python script for using focal length, disparity and baseline to calculate depth that may be a helpful reference if you have not seen it already.

https://github.com/IntelRealSense/librealsense/blob/master/doc/depth-from-stereo.md

The script only uses the fx value, likely because the values of fx and fy are often close to each other.