realsenseai/librealsense #10576: trigger number mismatch between depth frame and color frame

ping28198, 2022-06-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | V2.50 | 
| Operating System & Version |   Ubuntu1804 arm  | 
| Kernel Version (Linux Only)    |  4.9.201-tegra    | 
| Platform                                 |  NVIDIA Jetson   |
| SDK Version                            | 2.50                          | 
| Language                            | cpp                       | 
| Segment			|  others                           | 

### Issue Description
I have two cameras, set up hardware synchronization to trigger the two cameras, and get depth frames and color frames. 
Both cameras use external triggers, and their external trigger pins are connected to Jetson's GPIO. 
Jetson's GPIO outputs pulses to trigger both cameras every 50 ms. Both cameras are connect with Jetson by two USB 3.2 gen1 ports( maybe from same one hub).
In the code, the parameter RS2_OPTION_INTER_CAM_SYNC_MODE is configured to 4, which is GENLOCK.  All resolution is set to 640x380. 
I do get the data, but the frame number of the color frame I get through the get_frame_number () function usually does not match the frame number of the depth frame. 
The color frame number is stable and consistent with the trigger counter. 
However, the  depth frame number is usually smaller than that of the color frame , and the difference becomes larger and larger with time. 
                  trigger count number
                 |
trigger   :235 - 1654782342111
serial:146222254945, time stamp: 1654782342132.2, 1654782342127.3, match:4.9, id delay:-230, trigid:235 - 225, trigger delay:32 
serial:146222250855, time stamp: 1654782342133.3, 1654782342127.5, match:5.9, id delay:-231, trigid:235 - 234, trigger delay:33 
                                                       |                             |                                                                                |        |
                                                       |                             |                                                                                |        depth frame number
                                                       |                             |                                                                                color frame number
                                                       |                             depth frame time stamp
                                                       color frame time stamp
  
This leads to a question: in the current framesets, the frame number of the depth frame does not match the color frame number. Is the depth frame the current frame? Or the history frame, the current depth frame is still in the queue?

MartyG-RealSense, 2022-06-09:
Hi @ping28198  Depth and RGB sync is achieved by using '3' as the Inter Cam Sync Mode value, which is the **Full Slave** mode (compared to '2' which is the depth-only Slave mode).

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#appendix-a

ping28198, 2022-06-10:
But the trigger frequency I want is 20Hz, which is not in the standard frequency. If the trigger mode is set to 3 and I need the standard frequency, what should I do?

closed 2022-06-22 by MartyG-RealSense