realsenseai/librealsense #10577: poll_for_frames() use case

CherifB98, 2022-06-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Operating System & Version |   {Ubuntu 20.04}   | 
| Language                            |  {python}                          | 

### Issue Description
I'm looking up to code with both methods wait_for_frames and poll_for_frames and compare them as I want to have under 1/FPS loop treatement time. 
Until now, I've been using wait_for_frames () to get the frames but streaming with an FPS = 30, I've found out that it takes at least 31 ms for acquiring a frame. I'm looking for a quicker method to use to gain more time. And I've found that poll_for_frames() is an alternative. Althought, I still cannot find an example illustrating how to use it.
Replacing wait_for_frames() by poll_for_frames() get me the error "null pointer passed for argument "frame_ref"". 
Any help about this ? Thanks !

MartyG-RealSense, 2022-06-09:
Hi @CherifB98  The **poll_for_frames()** instruction will typically be used in multiple camera applications, such as the RealSense SDK's C++ **rs-multicam** multiple camera example program.

https://github.com/IntelRealSense/librealsense/blob/master/examples/multicam/rs-multicam.cpp#L49

A Python demonstration of multicam code that uses poll_for_frames() is provided in the **realsense_device_manager.py** script that is part of the SDK's 'box_dmensioner_multicam' example project for measuring the volume of a box.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/realsense_device_manager.py#L221-L222

A complication with poll_for_frames() compared to wait_for_frames() is the recommendation to control when the CPU is put to sleep and for how long in order to avoid maxing out the CPU usage percentage at 100%.

Otherwise, if you are using a single camera in your application then wait_for_frames() should be sufficient.  A RealSense team member describes the difference between the instructions at https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709

CherifB98, 2022-06-10:
Thank you @MartyG-RealSense for the clear response. 
I am indeed using multiple camera system. 
Do you have any solution for the error I am getting ("null pointer passed for argument "frame_ref"") ? 
I have searched on the recent issues but none of them helped me.

MartyG-RealSense, 2022-06-10:
A Python user at https://github.com/IntelRealSense/librealsense/issues/5903 who experienced the **null pointer passed for argument "frame_ref"** error when using poll_for_frames() was advised by a RealSense team member to use the SDK's **Keep()** instruction, which stores frames in memory.

More information about using Keep() in Python is at https://github.com/IntelRealSense/librealsense/issues/9251#issuecomment-864434816

MartyG-RealSense, 2022-06-16:
Hi @CherifB98  Do you require further assistance with this case, please?  Thanks!

CherifB98, 2022-06-16:
Hi @MartyG-RealSense ! I abandoned using poll_for_frames and sticked with wait_for_frames ! Thank you for your help