realsenseai/librealsense #10578: how to declare the temporal filter object

slippin-jimm, 2022-06-09:
I was trying to use the temporal filter on my code besides the hole filling filter but  when I declare one, the other stops working.  I have them both on the same function for getting the frames.

davidmartinez13, 2022-06-09:
I encountered this problem already. I basically defined a class where the filtering objects were created with the constructor and outside the main function to get frames. Hope it helps.
`    
class DepthCamera:

    def __init__(self):
        # Configure depth and color streams
        self.pipeline = rs.pipeline()
        config = rs.config()
        # Get device product line for setting a supporting resolution
        pipeline_wrapper = rs.pipeline_wrapper(self.pipeline)
        pipeline_profile = config.resolve(pipeline_wrapper)
        device = pipeline_profile.get_device()
        device_product_line = str(device.get_info(rs.camera_info.product_line))
        config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
        config.enable_stream(rs.stream.color, 1920, 1080, rs.format.bgr8, 30)
        # Start streaming
        profile = self.pipeline.start(config)
        depth_sensor = profile.get_device().query_sensors()[0]
        depth_sensor.set_option(rs.option.enable_auto_exposure, False)
        rgb_sensor = profile.get_device().query_sensors()[1]
        rgb_sensor.set_option(rs.option.enable_auto_exposure, False)
        self.temporal = rs.temporal_filter()
        self.hole_filling = rs.hole_filling_filter()
        self.hole_filling.set_option(rs.option.holes_fill, 2)

    def get_frame(self):
        frames = self.pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        depth_frame = self.hole_filling.process(depth_frame)
        # depth_frame = self.temporal.process(depth_frame)
        colorizer = rs.colorizer()
        colorizer.set_option(rs.option.color_scheme, 0)
        colorized_depth = np.asanyarray(colorizer.colorize(depth_frame).get_data())
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        if not depth_frame or not color_frame:
            return False, None, None, None
        return True, depth_image, color_image, colorized_depth

    def release(self):
        self.pipeline.stop()`

MartyG-RealSense, 2022-06-09:
@davidmartinez13  Thanks very much for your contributions!

@slippin-jimm  Please also check that you have listed the hole-filling filter **after** the temporal filter, as demonstrated in the script of @davidmartinez13 in the link above and in Intel's recommended listing order for the post-processing filters at the link below.

https://dev.intelrealsense.com/docs/post-processing-filters#using-filters-in-application-code

MartyG-RealSense, 2022-06-09:
It's great to hear that the solution provided by @davidmartinez13 worked for you - thanks very much for the update!