realsenseai/librealsense #10581: When to expect librealsense2-dkms support for kernel 5.13 eg.  ubuntu 20.04 hwe kernels

hernot, 2022-06-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 20.04) | 
| Kernel Version (Linux Only)    |   5.15.0-33                                       | 
| Platform                                 | PC  |
| SDK Version                            |  recent apt upgrade 2.x                           | 
| Language                            |  not relevant                          | 
| Segment			|  others                           | 

### Issue Description
When to expect librealsense2-dkms  support for recent ubuntu 5.13 hwe kernel?
Would there be any obstacles to be expected which are to be treated by librealsense team beyond changing config file to state that 5.13 is supported?

MartyG-RealSense, 2022-06-10:
Hi @hernot  I am not aware of future plans to update the kernel versions supported by the DKMS packages, which currently officially support up to kernel 5.4. 

[image]

A source-code build of the SDK supports newer kernels (5.8 and 5.11).  Unsupported kernels can be used with the SDK but there may be unpredictable consequences in regards to stability.  

Alternatively, if the SDK is built from source code with the **RSUSB backend** installation method then it bypasses the kernel and so is not dependent on Linux versions or kernel versions and does not require patching.

The patch that added kernel 5.11 support was contributed by a RealSense user via a GitHub fork of librealsense.

https://github.com/kevindehecker/librealsense/tree/patch_ubuntu_focal_5.11

In their GitHub fork, it looks as though they provided 5.11 support by editing the script file 'patch_utils.sh' and adding a script file called 'realsense-camera-formats-focal-hwe-511.patch'

https://github.com/kevindehecker/librealsense/tree/patch_ubuntu_focal_5.11/scripts

Conceivably you could try following their 5.11 method to be able to build the SDK from source code with 5.13 support, though I understand that this will not solve the wish for 5.13 DKMS package support.

MartyG-RealSense, 2022-06-17:
Hi @hernot  Do you require further assistance with this case, please?  Thanks!

hernot, 2022-06-17:
Not so far thanks for the update

closed 2022-06-17 by hernot

MartyG-RealSense, 2022-06-17:
You are very welcome.  :)