realsenseai/librealsense #10583: Build pyrealsense2 from source does not work.

remilamoureuxlevesque, 2022-06-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |  Linux 18.04 LTS  | 
| Kernel Version (Linux Only)    |  4.9.253-tegra                                         | 
| Platform                                 | NVIDIA Jetson Nano developpment kit |
| SDK Version                            | 2.50.0                      | 
| Language                            |  python                          | 
| Segment			|  Embedded                         | 

### Issue Description
I try to install pyrealsense2 from [this](https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-665650908) guide, but I got this issue : 

[image]

I also update cmake to the latest version (3.23.2).
Try to run the cmake command from the librealsense folder without the ../ and still got the same result.

MartyG-RealSense, 2022-06-10:
Hi @remilamoureuxlevesque  For a pyrealsense2 installation on Nano I recommend the Nano-specific guide at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 that a RealSense user created a few months after the instructions that you used were published.

remilamoureuxlevesque, 2022-06-11:
Hi @MartyG-RealSense, I try the guide from #6964 but for me I did not work, the build process did not succedm, but I find [this](https://github.com/IntelRealSense/librealsense/issues/7722#issue-736687248) guide that work for me.
Now I can import pyrealsense2 in my project.

Thanks for your help, it is very appreciate.

closed 2022-06-11 by remilamoureuxlevesque

MartyG-RealSense, 2022-06-11:
You are very welcome, @remilamoureuxlevesque - I'm pleased to hear that you achieved a solution.  Thanks very much for the update!