realsenseai/librealsense #10584: Depth Quality Tool Crashing

Cpeck71, 2022-06-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515  | 
| Firmware Version                   | 01.05.08.01| 
| Operating System & Version | Linux Ubuntu 18.04 | 
| Kernel Version (Linux Only)    | 4.9.253-tegra | 
| Platform                                 | NVIDIA Jetson Nano  |
| SDK Version                            |  2.50.0                          | 
| Language                            |  ~                         | 
| Segment			|  Depth Tool                          | 

### Issue Description
Howdy Realsense Team,

I'm trying to open the Depth Quality tool, however my computer (Jetson Nano, 4 GB RAM) crashes when it's loading. It doesn't get beyond opening the window for the tool, so I don't ever see any warnings etc. I suspect it might be a RAM issue or other hardware problem but I figured I'd reach out to see if there was a way to diagnose the problem. Looking at previous cases, i saw #4948 but I'm not sure I'm having the same version issue.

Thanks!

Cpeck71, 2022-06-10:
Hi Marty,

Yes, the viewer works perfectly fine

MartyG-RealSense, 2022-06-10:
If a camera can work normally on the Viewer then it usually works on the Depth Quality Tool as well.

Please try unplugging the micro-sized end of the USB cable from the base of the L515 camera, turning it around the opposite way and re-inserting it (USB-C cables are two-way insertion at the micro-sized end).  If it does not resolve the problem with launching the DQT then change the connector back to its previous orientation.

MartyG-RealSense, 2022-06-17:
Hi @Cpeck71  Do you require further assistance with this case, please?  Thanks!

Cpeck71, 2022-06-17:
Hi Marty, I'm sorry I've been busy with another task recently and haven't been able to attend to the depth quality in the past week.  I should have time next week to pick it back up, can I keep the issue open in the meantime or should I close it and re-open when I have progress? Whatevers best for y'all.

Thanks!

MartyG-RealSense, 2022-06-17:
If you will likely be picking it up again next week then it's fine to keep it open.  You are welcome to close it yourself later if you find that it is going to be longer before you return to it.  Good luck!

Cpeck71, 2022-06-23:
Hi @MartyG-RealSense I'm back! So i just flipped the usb cable but my computer still crashed. (i put it back the other way too)

closed 2022-06-23 by Cpeck71

MartyG-RealSense, 2022-06-23:
That's excellent news that you achieved a positive outcome.  Thanks very much for the update  :)