realsenseai/librealsense #10586: Cannot start Realsense Viewer

Lasweiga, 2022-06-12:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:572

```

MartyG-RealSense, 2022-06-13:
Hi @Lasweiga  There was a recent case of this error on Windows at https://github.com/IntelRealSense/librealsense/issues/10544#issuecomment-1138569631 that was resolved with the following steps.

1.  Download and run the standalone **2.21.0** version of the RealSense Viewer (**Intel.RealSense.Viewer.exe**) from the 'Assets' file list at the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.21.0

Once downloaded, Intel.RealSense.Viewer.exe should be launched with a double-leftclick from the folder location on your computer that it was downloaded to and not the Viewer launch shortcut icon on the Windows desktop.

2.  If the 2.21.0 version of the Viewer has launched successfully, close it and then launch the **2.50.0** version of the Viewer with the usual desktop launch icon.  The previous error should now be gone and the Viewer can launch normally.

Lasweiga, 2022-06-17:
@MartyG-RealSense . Thank you the info. I have done the steps but did not fix the issue. at all . The same error poped up when I  launched the realsense viewer v 2.50. Earlier version was running fine.
What would be the next step to resolve this? Thanks.

MartyG-RealSense, 2022-06-17:
Thanks for testing.  Let's try the standalone Intel.RealSense.Viewer.exe from 2.49.0 next to check whether or not the problem is with the 2.50.0 Viewer specifically.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.49.0

[image]

MartyG-RealSense, 2022-06-23:
Hi @Lasweiga  Do you require further assistance with this case, please?  Thanks!

Lasweiga, 2022-06-23:
Hi Marty,

Sorry for the late reply. Unfortunately, the 2.49.0 version was behaving
exactly the same. See screenshots attached. I also made sure I got the
latest graphics driver installed, in case  problems with opengl, but did
not make any difference.

MartyG-RealSense, 2022-06-24:
Is the **rs-capture** example program, which also makes use of OpenGL graphics, able to launch please?  Instructions for launching this program on the Windows version of the RealSense SDK can be found at https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656

MartyG-RealSense, 2022-06-30:
Hi @Lasweiga  After reading the comment above, do you require further assistance with this case, please?  Thanks!

closed 2022-07-12 by MartyG-RealSense