realsenseai/librealsense #10587: rs2_load_json error, device_type key is not supported

JACKLiuDay, 2022-06-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {ubuntu18.04  | 
| Kernel Version (Linux Only)    |  (4.9.140)                                         | 
| Platform                                 |  NVIDIA Jetson NX jetpack 4.4.1  |
| SDK Version                            |  { 2.43.0 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {}                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
[image]
When I load json file with realsense-viewer. It shows error.
[image]

MartyG-RealSense, 2022-06-13:
Hi @JACKLiuDay  This appears to be a json containing IMU calibration data.  The RealSense Viewer tool does not import this type of json file.  Instead, the Viewer can save and load **Visual Preset** jsons that define values for option settings that affect the depth, infrared and RGB images.

https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets

JACKLiuDay, 2022-06-13:
> Hi @JACKLiuDay This appears to be a json containing IMU calibration data. The RealSense Viewer tool does not import this type of json file. Instead, the Viewer can save and load **Visual Preset** [...]

Hi, thank you so much for your reply. I run the rs-imu-calibration. While, it shows error when writing calibration to device. So I tried to load the calibration json file with realsense-viewer. 
[image]

MartyG-RealSense, 2022-06-13:
I hope that the advice in a Jetson NX case at https://github.com/IntelRealSense/librealsense/issues/9857 that had the same problem will be helpful to you in resolving it.

JACKLiuDay, 2022-06-13:
> I hope that the advice in a Jetson NX case at #9857 that had the same problem will be helpful to you in resolving it.

[image]
It worked! Thank you so much!

closed 2022-06-13 by JACKLiuDay

MartyG-RealSense, 2022-06-13:
You are very welcome.  That's great news.  :)