realsenseai/librealsense #10590: multiple D435i cameras stuck at frame 15

Lawrencewzx, 2022-06-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |   {Win 10  | 
| Platform                                 | PC  |
| SDK Version                            | 2.49.0 }                          | 
| Language                            |  {python 3.8 }                          | 
| Numpy                            |  {1.22.4 }                          | 

### Issue Description
hi,  
When I use more than two D435i cameras to collect pictures at the same time, the first 15 frames can be collected normally, but the RGB images after that do not change and return the same picture all the time.  Depth images can be collected normally.  I had a similar problem with a D435i camera and couldn't get any frames after 15 frames.  Again, 15 frames.
@[MartyG-RealSense](https://github.com/MartyG-RealSense)
The following code

  `
  class Realsense2:
        def __init__(self, camera_id_list=[0], camera_width=640, camera_height=480, camera_fps=60):
            self.camera_width = camera_width
            self.camera_height = camera_height
            self.camera_fps = camera_fps
            self.camera_id_list = camera_id_list:

          def camera_config(self):
              self.connect_device = []
              # get all realsense serial number
              for d in rs.context().devices:
                  print('Found device: ',
                        d.get_info(rs.camera_info.name), ' ',
                        d.get_info(rs.camera_info.serial_number))
                  if d.get_info(rs.camera_info.name).lower() != 'platform camera':
                      self.connect_device.append(d.get_info(rs.camera_info.serial_number))
              # config realsense
              self.pipeline_list = [rs.pipeline() for i in range(len(self.camera_id_list))]
              self.config_list = [rs.config() for i in range(len(self.camera_id_list))]
              if len(self.camera_id_list) == 1:  # one realsense
                  self.config_list[0].enable_device(self.connect_device[0])
                  self.config_list[0].enable_stream(rs.stream.depth, self.camera_width, self.camera_height, rs.format.z16,
                                                    self.camera_fps)
                  self.config_list[0].enable_stream(rs.stream.color, self.camera_width, self.camera_height, rs.format.bgr8,
                                                    self.camera_fps)
                  self.pipeline_list[0].start(self.config_list[0])
              elif len(self.camera_id_list) >= 2:  # two realsense
                  if len(self.connect_device) < 2:
                      print(
                          'Registrition needs two camera connected.But got one.Please run realsense-viewer to check your camera status.')
                      exit()
                  # enable config
                  for n, config in enumerate(self.config_list):
                      config.enable_device(self.connect_device[n])
                      # config.enable_stream(rs.stream.depth, self.camera_width, self.camera_height, rs.format.z16,
                      #                      self.camera_fps)
                      config.enable_stream(rs.stream.color, self.camera_width, self.camera_height, rs.format.bgr8,
                                           self.camera_fps)
                      self.config_list[n] = config
                  # start pipeline
                  for n, pipeline in enumerate(self.pipeline_list):
                      pipeline.start(self.config_list[n])
      
          def wait_frames(self, frame_id=None):
              '''
              camera_id:
                  @ = camera number , get all frame
                  @ = id , get specific id camera frame
              '''
              self.frame_list = [None for i in range(len(self.camera_id_list))]
              if frame_id != None:  # give a frame id
                  self.frame_list[n] = self.pipeline_list[frame_id].wait_for_frames()
              else:  # get all frame
                  if len(self.camera_id_list) == 1:
                      self.frame_list.append(self.pipeline_list[0].wait_for_frames())
                  else:
                      for n, camera_id in enumerate(self.camera_id_list):
                          self.frame_list[n] = self.pipeline_list[n].wait_for_frames()
                          self.frame_list[n] = align.process(self.frame_list[n])
      
          def rgb_image(self, camera_id):
              color_frame = self.frame_list[camera_id].get_color_frame()
              color_image = color_frame.get_data()
              return color_image
      
          def depth_frame(self, frames):
              depth_frame = frames.get_depth_frame()
              return depth_frame
      
           def depth_frame(self, camera_id):
               depth_frame = self.frame_list[camera_id].get_depth_frame()
               depth_frame = np.asanyarray(depth_frame.get_data())
               depth_image_3d = np.dstack((depth_frame, depth_frame, depth_frame))
               depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image_3d, alpha=0.03), cv2.COLORMAP_JET)
               return depth_colormap
      
          def stop(self):
              for pipeline in self.pipeline_list:
                  pipeline.stop()
              print("camera exit sucessfully.")
`

MartyG-RealSense, 2022-06-14:
Hi @Lawrencewzx  When a pyrealsense2 program stops after 15 frames, it is usually because an **append** instruction has been used, as it has been in your script.  A solution for this is to store frames in the computer's memory with the **Keep()** instruction.  More information about this can be found at https://github.com/IntelRealSense/librealsense/issues/9251#issuecomment-864434816

Lawrencewzx, 2022-06-14:
oh, thank you
yeah, i'm using two D435i. But we'll probably use more later.

Lawrencewzx, 2022-06-14:
> Hi @Lawrencewzx When a Python program in the RealSense SDK stops after 15 frames, it is usually because an **append** instruction has been used, as it has been in your script. A solution for this is to store frames in the computer's memory with the **Keep()** instruction. More information about this can be found at [#9251 (comment)](https://github.com/IntelRealSense/librealsense/issues/9251#issuecomment-864434816) though I note that you have tried Keep() already and are also able to store depth frames successfully, the problem being with the RGB frames.
> 
> It also sounds as though your code works with 2 cameras but not more than 2. Is this correct, please?
> 
> Please try changing the FPS from 60 to 15 to test whether more than 2 cameras might be generating too much data for the computer or the program to handle. The more cameras that are enabled, the more of the computer's resources that are consumed. For example, a computer with an Intel i7 processor would be recommended for 4 cameras that are active simultaneously on the same computer.

my cpu is  i5-11400，that should do the trick.
i changed the FPS from 60 to 30, 15, that's always been the prpblem

Lawrencewzx, 2022-06-14:
Not just at 30 frames, 15FPS, 60FPS  all  turn the same problems:  the same  images after 15 frames
I still haven't dealt with the problem with the keep() function, can you give me a more specific way to deal with it?

Lawrencewzx, 2022-06-14:
> Not just at 30 frames, 15FPS, 60FPS all turn the same problems: the same images after 15 frames
> I still haven't dealt with the problem with the keep() function, can you give me a more specific way to deal with it?

Not just at 30 frames, 15FPS, 60FPS all turn the same problems: the same images after 15 frames
I still haven't dealt with the problem with the keep() function, can you give me a more specific way to deal with it?

Lawrencewzx, 2022-06-14:
> Yes, that CPU should be very sufficient. As the advice about an i7 for 4 cameras was given four years ago, you may even be able to run 4 cameras off your modern 11th generation i5.
> 
> So the problem still occurs at 30 FPS?

Not just at 30 frames, 15FPS, 60FPS all turn the same problems: the same images after 15 frames
It is worth mentioning that I was able to capture depth images without any problems before using the keep () function.  That said, IF I do it right, I don't think the keep () function really works, which is where I get confused 

I still haven't dealt with the problem with the keep() function, can you give me a more specific way to deal with it?

MartyG-RealSense, 2022-06-14:
A limitation with Keep() to bear in mind is that because it stores frames in memory, the recording period is determined by the memory capacity of the particular computer that the script is being used on, because memory is progressively consumed by the stored frames over time.  So a computer with an average specification may be able to store 10 seconds worth of frames, whilst a computer with a larger memory capacity may be able to store 30 seconds worth.

I note that your script is using **wait_for_frames()**.  This is not recommended by Intel for multiple camera scripts where the same application is accessing more than one camera.  It is recommended that **poll_for_frames()** is used instead.  A RealSense team member describes the difference between wait_for_frames() and poll_for_frames() and the reason for using poll_for_frames() in multicam applications at https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709

At https://github.com/IntelRealSense/librealsense/issues/8388#issuecomment-782395443 a RealSense user shared a Python script that uses append but not Keep().  Depth and color from all attached cameras is streamed and aligned and displayed in individual windows, and a simultaneous capture to .png image file of depth and color from each camera is made whenever the 's' key is pressed.

closed 2022-06-26 by MartyG-RealSense