realsenseai/librealsense #10594: Big error after calibration

Obrepal, 2022-06-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |   Win 10/MacOS  | 
| Platform                                 | PC|
| SDK Version                            |  { legacy / 2.50.0}                          | 
| Language                            |  {python}                          | 

RS was calibrated with on Chip and dynamic tool (with RGB). I used my phone Redmi Note 8T. 
Result in chip was 0.00 which is the best result one can obtain thanks to pattern provided here (https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras). During the dynamic part, I didn't change distance between pattern and camera plane. 
Also, phone was kept straight and whole scene was covered (15 pictures were spread).
I tried it multiple times and each of the results is horrible. Below I put comparison of d435 and d435i (1000MM). 
<img width="96" alt="image" 
src="https://user-images.githubusercontent.com/48407260/173640994-61e56839-addf-45a8-9e1d-41f1915cee7d.png">.
 

As far as I know, RMS should be close to 9 MM (according to Intel).

MartyG-RealSense, 2022-06-14:
Hi @Obrepal   Could you confirm a couple of points about the target image that you were using, please?

1.  If you were using a target on an Android phone with the Dynamic Calibration tool, did you use the Android target app on the Google Play Store at the link below that the user guide for the tool recommends?

https://play.google.com/store/apps/details?id=com.intel.realsenseviewer17613&gl=GB

2.  The On-Chip calibration tool is used to improve depth image quality.  If you are aiming to improve depth accuracy then the **Tare calibration** tool should be used instead.  This has a different printable target image to the On-Chip calibration tool that can be downloaded at the link below in two different paper sizes.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#ground-truth-targets-and-printing--mounting-instructions

[image]

**********

You could also try using a tool built in to the RealSense Viewer to reset the camera to its factory-new default calibration to see whether this provides the desired image quality / accuracy.  Instructions for doing so can be found at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

Obrepal, 2022-06-15:
Yes, I downloaded app via google store (Dynamic Target tool version 1.0)

Ok, should I print it or it can be displayed on monitor? 
Also how accurate it has to be to improve performance (I do not have laser just tape measure)?

MartyG-RealSense, 2022-06-15:
The calibration bar image provided by the Android app should be displayed on the screen of the Android device and not printed.  The camera should be pointed at the image on the Android screen.

Aside from using the Dynamic Calibration tool in a location with good lighting, there is not much else that can be done to improve its performance further.

The Tare calibration tool offers more flexibility, as it has a **Get** button to custom-adjust the depth calibration for the distance of the camera from the observed surface before clicking the 'Calibrate' button to begin calibration.

[image]

Obrepal, 2022-06-15:
Yeah, in Dynamic I used screen of phone.

Ok, I will try it and let you know.
Thanks :)

MartyG-RealSense, 2022-06-21:
Hi @Obrepal  Do you require further assistance with this case, please?  Thanks!

Obrepal, 2022-06-21:
Hi @MartyG-RealSense I think it was improved results are not as good as expected but it will do it. 
[image]

MartyG-RealSense, 2022-06-21:
Comparing the estimated RMSE curve for D435 on Intel's official graph from their camera tuning guide to your own graph, they seem to be similar at the 4000 mm distance.  Intel's graph is clearly marked as 60 mm RMSE at 4000, whilst your own curve looks to be around 60 at that distance if a ruler is put over the chart at the halfway poit between 0 and 100.

[image]

Obrepal, 2022-06-21:
I am using 640x480 insteas of 848x480 thus the difference

MartyG-RealSense, 2022-06-21:
In general, as resolution is increased the depth accuracy increases, and accuracy decreases as resolution is reduced.  The optimal depth accuracy resolution for the D435i camera model is 848x480.  I do not believe that there will be a huge reduction in depth accuracy at 640x480 though.

RMS error may reduce if the camera's IR emitter and its dot pattern projection are disabled.  If the scene that the camera is observing is well lit then the camera can use the ambient light in the scene to analyze surfaces for depth detail instead of analyzing the dot pattern.

closed 2022-07-03 by MartyG-RealSense