realsenseai/librealsense #10595: Why intrinsic parameters for D455 RGB are changed

m3sen, 2022-06-15:
| Camera Model                       | D455          | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Win10       | 
| Platform                                 | PC               |
| SDK Version                           |  v2.50.0       | 
| Language                               |  C++           | 
| Segment			        |  others         | 

### Issue Description
Why intrinsic parameters for RGB are changed? when I just execute realsense viewer.

MartyG-RealSense, 2022-06-15:
Hi @m3sen  Are you referring to the color intrinsic **coefficients** being non-zero values when the other intrinsic types such as depth and infrared have all-zero coefficients.  If so then this is normal on D455.  Usually the coefficients are all zero, but a rare exception is the color coefficients on the D455 model specifically which are non-zero.  It is not something to be concerned about.

m3sen, 2022-06-15:
Thank you @MartyG-RealSense
I have been debugging the problem and then   symptom is as like following.
Why RGB instrinsic parameter is changed?
Is OCC performed?
## Symptom reproduction
### usb dump with wireshark when GETINTCAL is excuted. 
### realsense viewer execute
[image]
### rs-align execute
[image]
### realsense viewer execute
[image]

MartyG-RealSense, 2022-06-15:
Thanks very much for the images from rs-enumerate-devices, that helps a lot.

The highlighted values **fx** and **fy** describe the focal length of the image as a multiple of pixel width and height.  The section of the SDK's **Projection** documentation at the link below provides further information.

https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20#intrinsic-camera-parameters

It looks as though the field of view (FOV) is also varying in your images as well as the focal length.  The link above explains that the FOV of an image may vary.  At https://github.com/IntelRealSense/librealsense/issues/4179#issuecomment-500860874 a RealSense team member how the FOV can be calculated using focal length.  

When performing depth-color alignment, the SDK's **align processing block** will automatically adjust the field of view sizes so that one matches the other, and also automatically adjusts for differences in resolution.  If depth to color alignment is performed then the depth FOV is resized to match the color FOV size.  Conversely, if color to depth alignment is performed then the color FOV is resized to match the depth FOV size.

As you are using **rs-align**, it therefore seems logical that variations in the values would be present before and after.  

My recommendation would therefore be to not be concerned about changes in the values when using rs-align.

closed 2022-06-16 by m3sen