realsenseai/librealsense #10596: Realsense d455 "auto exposure burnout" after running a while on 1280x720 depth resolution.

MartinPedersenpp, 2022-06-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D455 | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |   {Win 10  |                                       | 
| Platform                                 | PC.  |
| SDK Version                            |  2.49.0                         | 
| Language                            |  python }                          | 
| Segment			|  others                        | 

### Issue Description
Hello

I am experiencing a "weird" issue where if I use my D455 camera with depth resolution 1280x720x30fps it will start "burning out" the auto exposure algorithm and in the end the result is unusable.
[image]

After unplugging and replugging the camera it works again:

[image]

I have tried running calibration and firmware resets, but I fear that it might be a hardware problem?

Please request if any more information is needed
Any help is much appreaciated.

Best regards
Martin

MartyG-RealSense, 2022-06-15:
Hi @MartinPedersenpp  It does not look as though it is a hardware fault.  Comparing the two images, I note that your ASIC temperature is 45 degrees C on the bad depth image and 42 degrees C after the reset.  I consider 42 to be a maximum operating temperature 'tipping point', above which there is the possibility of problems manifesting in the image.

Factors that could cause a high operating temperature are discussed in another currently open case at https://github.com/IntelRealSense/librealsense/issues/10574#issuecomment-1155390920

MartinPedersenpp, 2022-06-15:
👍🏼 makes sense since all but the sensor side of the camera is mounted in a enclosed space. I will keep track of the other post. Thanks for pointing it out.

closed 2022-06-15 by MartinPedersenpp